gregsaun/prusa_i3_bear_upgrade #1: measurements, Y axis

zagarolo, 2017-10-05:
Hi his beautiful compliment project continue and I'm here better with her !!

I bought all the prices for assembling the Y axis ...

I have some questions:

the original MK2S pria fit the 100mm Y axis instead of its 110mm V-Slot design and 185mm later, its 195mm V-Slot design ever? Bearing Y bearings may have length problems

excuse my English to use the translator

gregsaun, 2017-10-05:
Thank you so much for the compliments, this is really helping me!

On the original MK2(s) those measurements are done from top of the rod holder, not from of the rod hole. It is difficult to explain, hope you understand. You can also have a look on this drawin for the MK2 : https://raw.githubusercontent.com/PrusaMK2Users/MK2_Tips_and_Tricks/master/Prusa-i3-mk2-Dimensions.png

If you carefully check the rod holders, they are just a bit bigger than the 20mm (width of the vslot), this is to have the exact same dimensions as the MK2(s) (rod hole is 9.75 long and between front of the hole and front of the piece you have 9.50).

However, I did two different piece from back and front and that is wrong. I didn't remember why I did this and there is effectively an issue! Thanks for that, will fix it today :)

gregsaun, 2017-10-05:
So I remember why I did different front and back rod holders. This is due to the fact the frame is not 6mm thick but more 6.2-6.3mm on mine. So, on the original, you have 185 only if you have a frame that is 6mm, otherwise you will compensate the difference by screwing a bit more on the rod.

I will make all rod_holders similar and the difference will be compensated when screwing in the vslot.

gregsaun, 2017-10-05:
01bf8536e8f9c28e50426bd4b3f99f8dfb1a0566

closed 2017-10-05 by gregsaun