gregsaun/prusa_i3_bear_upgrade #3: Questions from Zagarolo

gregsaun, 2017-10-07:
> hi thank you for your reply...
> 
> you add print setting is for all parts or only psu and holder mount ? thanks

> drill new holes print 3d for frame Z ?

gregsaun, 2017-10-07:
I am currently testing all prints and once I have the expected results I add it to the print settings. All pieces will be in as soon as I can ;)

I do not understand what you means by "drill new holes print 3d for frame Z ?"

zagarolo, 2017-10-07:
I have printed all the parts, should I print everything?

yes, holes for Z axis

Can a print be positioned above to help for the holes?

zagarolo, 2017-10-07:
sorry 

Have I printed all the parts, should I reprint everything?

zagarolo, 2017-10-07:
wait for new settings to print?

gregsaun, 2017-10-08:
If your print settings are stronger than what I propose, then you do not necessarily need to reprint them. However, I am still working on each printed parts and solve details and sometime serious issues like the psu mounts. You might have issues when using the parts you have printed it. If that is the case, check if there is an update and if not you can raise a bug here.

Here are the dimensions for the Z axis holes : 
https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/master/doc/frame_drilling_dimensions.pdf

I have done a plate but never tested it, so use it with caution (and please report feedback) : 
https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/master/printed_parts/stl/drilling_helper.stl

zagarolo, 2017-10-08:
Hi 

I read a comment where to ask for help to complete the list of the pieces as never are they all?

zagarolo, 2017-10-08:
I've already bought all the pieces ready to be cut

zagarolo, 2017-11-01:
Hi I mounted all the printer I inserted igus new bearings and rods the printer works but Z wobble even with the original rods and bearings, I confirm to modify psu mount 1 mm even if very fixed, another problem the rod clamps come in in the cart. the Zwobble same thing with the new Z tops you've designed in optional parts.
if you have an email I'll send you the photos

gregsaun, 2017-11-01:
Hi Zagarolo, it should not Z wobble, the only flex you can have might come from the alu frame. I got another feedback and it was clearly sturdier and no wobble.

I do not understand what do you mean by "another problem the rod clamps come in in the cart".

For the images, you can use image hosting services and use the following code in your comment : 
```markdown
![short description of image](image_url)
```

Did you use Openbuilds parts or other brands?

Some things to check :
* All screws
* Rods should not move in the rod_holders, it should be tight
* Threads made in the vslots to fix the frame
* Extruder calibration

zagarolo, 2017-11-01:
hi 

All screws = yes ok 
Rods should not move in the rod_holders, it should be tight = yes ok
Threads made in the vslots to fix the frame = yes ok 
Extruder calibration = yes ok 

problem zip fix rod asix X

zagarolo, 2017-11-01:
use Openbuilds parts

gregsaun, 2017-11-02:
Could you add pictures here?

Could you explain what do you mean by this two sentences : 
*"problem zip fix rod asix X"*
*"another problem the rod clamps come in in the cart"*

gregsaun, 2017-11-02:
Is this issue what you are talking about?
https://github.com/gregsaun/prusa_i3_bear_upgrade/issues/8

gregsaun, 2017-11-06:
I close this one as we continue on the other one :)

closed 2017-11-06 by gregsaun