gregsaun/prusa_i3_bear_upgrade #4: Not a bug: sources for extrusion

FischOderAal, 2017-10-08:
I've seen that you plan to cut everything yourself. Sometimes this and openbuilds (due to taxes) is not an option, so maybe it makes sense to list options where you can order the pieces pre-cut. For Germany good sources are:
http://www.motedis.com
http://kritec-gmbh.de

gregsaun, 2017-10-08:
Good idea, we can start listing few here. Please, anyone know a good one, please help us to complete the list :)

UK : ooznest.co.uk (very good)
Portugal : http://www.ratrig.com (not tested)

FischOderAal, 2017-10-11:
Some more:
Germany: http://www.watterott.com/de/Mechanik/V-Slot-Profile
Germany: http://www.conucon.de/de/29-openbuilds
France: http://www.vslot-europe.com/home/8-vslot-rail.html
Switzerland: https://www.3dware.ch/OpenBuilds-V-Slot-Profil-20mm-x-40mm-x-1000mm-De.htm

I also found this:
https://www.google.com/maps/d/u/0/viewer?mid=1pluH7gcGIW6e3wA9oDI7jkD65Ds&hl=en_US&ll=49.25723077144779%2C9.548680559375043&z=6

That was all I could find while going through a mug of coffee :)

gregsaun, 2017-11-06:
Forget to say thank you FischOderAal, sorry! Thank you then :)

I have added a page with online shops selling official Openbuilds hardware (and pre-cut option) : https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/master/doc/openbuilds_shop_list.md

DoItWithASmile, 2018-02-28:
This is more FYI:
I received my hardware from RatRig today. Here is something I really don't like with most shops listed:

The tee nuts are terrible!
[image]

This is how a proper tee nut should look like IMHO:
[image]

gregsaun, 2018-02-28:
They look like similar to Openbuilds tee nuts : 
[image]

I totally agree they are not the best one but they work pretty well and they are cheaper than "bigger" ones.

DoItWithASmile, 2018-02-28:
Found a shop in Algeria offering v-slot heavy duty nuts for 0.10 EUR per piece 😅 
www.dzduino.store

No pre-cutting of extrusions though.

gregsaun, 2018-03-01:
I did not realize yesterday but those type of tee nuts will enter in contact with my printed parts due to the extra material on top

[image]

However this type will be good (from dzduino.store) : 
[image]

dukemagus, 2018-03-01:
https://www.tdtec.com.br/

These guys sell 20x40 aluminum profiles on the openbuilds.com format in Brazil.

You'll need to contact them to be sure of which other related components they sell, though

gregsaun, 2018-03-01:
Thank you, I have added them to the list. tdtec.com.br does not work in my country, are you sure about the link?

Panayiotis-git, 2018-03-01:
Greece: https://grobotronics.com/mechanics-and-wheels/openbuilds-v-slot/

adam6666, 2018-03-03:
India - https://www.3dprintronics.com/Aluminium-Extrusions-c7124102 . They have V Slot extrusions and accessories but are not official V Slot distributors.

fishmsk, 2018-03-04:
Russia - https://soberizavod.ru/
They have aluminum extrusions, screws, tee nuts, plates etc. The only difference from Openbuilds' extrusions is a hole - it is for M6 thread here (Openbuilds ones have holes for M5 thread).

gregsaun, 2018-03-04:
Thank you all, we start to have a nice list :)

fishmsk > Do you mean the tee nuts are M6? Do they propose cutting service?

adam6666, 2018-03-04:
https://www.3dprintronics.com/Aluminium-Extrusions-c7124102 these guys provide precise customized lengths of all the T slots and V Slots.

fishmsk, 2018-03-05:
gregsaun -> Yes, they provide cutting service.
I mean not Tee nuts hole but the hole in the extrusion itself. Please see the drawings below:
[http://protein.nmr.ru/fish/pics/soberizavod_vs_openbuilds.jpg](url)

gregsaun, 2018-03-05:
Thanks  adam6666 and fishmsk.

I have just added new stores in Canada, Norway, Spain and Sweden.

DoItWithASmile, 2018-03-15:
Denmark: https://3dnielsen.dk

Didn't check if they offer pre-cut.

DoItWithASmile, 2018-03-15:
They to offer pre cut with +/- 0.5 mm

gregsaun, 2018-03-15:
Thank you, it's in  :)

adam6666, 2018-03-21:
Hello Greg
There is a slight issue which I am facing. My printer is complete but during calibration, it shows left front calibration point can't be reached. Although when I check it manually, the pinda probe is exactly in the center of it. Please advise

gregsaun, 2018-03-21:
Hi adam, I have opened a new issue here : https://github.com/gregsaun/prusa_i3_bear_upgrade/issues/20

jhoff80, 2018-04-06:
Will the RatRig kits still need the joining plates to be filed down slightly?  I assume so, but just wanted to check before buying a tool for that.

gregsaun, 2018-04-06:
That is a good question because they are coming in black! No one report something about this and I do not have RatRig plates in hand so hard to give an precise answer. I will ask RatRig directly and come back to you ;)

gregsaun, 2018-04-09:
Got an answer from RatRig, plates are not sharp like original Openbuilds plates so no need to file them. I'll add a note in the manual. Thanks for reporting :-)

kberck, 2018-07-11:
Does the extruded aluminum have to be v-slot specifically?  I have some t-slot available, but it doesn't have the chamfered edge of the v-slot.  Wanted to check before I start cutting so I don't hit a roadblock later on.
Thanks!

gregsaun, 2018-07-12:
@kberck yep this is mandatory because of the way my parts are designed. Also check here : 
https://github.com/gregsaun/prusa_i3_bear_upgrade/issues/39

martinkaersgaard98, 2019-02-02:
To the list of sources of V-slot rail and other OpenBuilds parts you can add Danish shop: makersupplies.dk

sparkyman215, 2019-04-26:
The NA - Texas distributor: smw3d tries to redirect to hobby-fab.com, a parked website. Suggest removing from list

gregsaun, 2019-04-30:
@sparkyman215 thanks!

closed 2019-11-07 by gregsaun