gregsaun/prusa_i3_bear_upgrade #5: Having troubles with frame_drilling_dimensions.pdf

alomex21, 2017-10-25:
I printed with adobe acrobat reader with real size, which print the size of the paper (letter in my case) intead of the whole sketch, as the result of some of the distances doubled(center of the aluminum extrusion with the edge of the z frame, which measures 120mm instead of 60mm and 40 mm instead of 20mm). Any help on this?
Thanks
[image]

FischOderAal, 2017-10-25:
In the end, it is only a sketch. I guess it was scaled to emphasis the details. You should print the drill template.

gregsaun, 2017-10-25:
I think you are both right :)

I made this drawing on A3 paper, this could explain why it is 2 times bigger. Be careful when using the printed design, carefully check all dimensions.

gregsaun, 2017-10-25:
For your information, the problem is that I used a scale of 2:1 to make this drawing. I have fix it by adding the scale on the drawing. You should then print 2x smaller.

Thanks for your feedback  :)

commit 3d15455c (2017-10-25): switch to A4 paper. add scale and units. This closes #5
closed 2017-10-25 by gregsaun