gregsaun/prusa_i3_bear_upgrade #10: Thanks gregsaun

zagarolo, 2017-11-02:
[image]
[image]
[image]
[image]

gregsaun, 2017-11-02:
:heart_eyes:

So good to see this in real!

y_rod_holder will be fixed in priority

zagarolo, 2017-11-02:
look at my side printing to see it's a bit misaligned or Z wobling you have a tip

gregsaun, 2017-11-02:
Did you re-calibrate the printer (calibrate xyz)? Did you check your Z level again?

zagarolo, 2017-11-02:
yes, lined XYZ I did not re-align Z axis again

gregsaun, 2017-11-02:
This might comes from the y_rod_holder too. Let's fix this first.

zagarolo, 2017-11-02:
Ok

FischOderAal, 2017-11-02:
This looks great. Congrats!

gregsaun, 2017-11-03:
New rod_holder available :)

zagarolo, 2017-11-05:
[image]
[image]

gregsaun, 2017-11-05:
yip yip :)

Do you speak Italian? I understand a little bit so you can write in italian and I will answer in english. (my mother tongue is french and I understand a very little bit of german as well)

zagarolo, 2017-11-05:
Ciao si sono italiano

zagarolo, 2017-11-05:
Ho visto che hai modificato rod holder oggi stampare e proverò

gregsaun, 2017-11-05:
I have seen the "quasar vetri" in your picture ;)

Nice. Do you have another spool of orange? :grin:

zagarolo, 2017-11-05:
Siii bello ho finito ultima bobbina ,,😯😩

zagarolo, 2017-11-05:
[image]
E finita 😰😰😰😭😭😭😭

gregsaun, 2017-11-05:
In case of I am using MF-PETG Neon Orange from extrudr.eu which is very very close to Prusa orange.
https://extrudr.eu/collections/petg/products/extrudr-mf-petg-neon-orange

zagarolo, 2017-11-05:
Si Colere molto vicino a Prusa

gregsaun, 2017-11-05:
I have added feet. For now, they have same shape than MK2(s), like this you can reuse your (home made) dampers

gregsaun, 2017-11-09:
I finished first build yesterday, the black corner on aluminum plate very makes the build very hard to square/tight. After several test on how we can square them correctly I have decided to remove them. Did you use them?

zagarolo, 2017-11-09:
[image]
[image]
[image]
[image]

zagarolo, 2017-11-09:
ciao ordinato  MF-PETG Neon Orange  e stampato tutto cosa ti riferisci black corner on aluminum plate yes 

come nel tuo progetto la parte del motore installati in basso e la parte anteriore cioe lcd  installati in alto per non avere problemi con il carrello si bloccava

zagarolo, 2017-11-09:
Hello ordered MF-PETG Neon Orange and printed everything you say black corner on aluminum plate yes

as in your project the part of the engine installed at the bottom and the front ie lcd installed up to not have problems with the car locked

gregsaun, 2017-11-09:
Ok thank you, I will still remove them, easier and cheaper and does not impact rigidity.

For MF-PETG Neon Orange I print at 225°C and 80°C for bed. Fan I am still testing but at 30-50 for now...

zagarolo, 2017-11-09:
MF-PETG 240  for bed 90 is perfect linear

zagarolo, 2017-11-09:
Manufacturer datasheet is 240 bed 90

gregsaun, 2017-11-09:
Where did you get the datasheet, on their website is 205 and 80? https://extrudr.eu/collections/petg/products/extrudr-mf-petg-neon-orange

zagarolo, 2017-11-09:
https://www.3djake.it/extrudr/filamento-mf-petg-neon-arancione

zagarolo, 2017-11-09:
Cosa sono queste righe più grande durante la stampa ? I puoi aiutare  
[image]

zagarolo, 2017-11-09:
ABS 
[image]

gregsaun, 2017-11-09:
Thanks for the link, interesting!

This is due to the fact it is doing parts of the line, then go somewhere else and then finish the started ones. Then you have this joining line. I do not know how to solve it but some things you can check : 
- layer height
- extruder calibration (over extrusion?)
- extruder temp?

zagarolo, 2017-11-17:
All parts orange neon petg
[image]

gregsaun, 2017-11-17:
Sweet build! In case of I updated the lcd supports to avoid colliding to with rod holders.

Do you still have the wobbling from the previous version?

zagarolo, 2017-11-17:
[image]
Hello no wobbling

zagarolo, 2017-11-17:
[image]

gregsaun, 2017-11-17:
Cool! Do you find it sturdier? Do you see a difference in print quality?

zagarolo, 2017-11-17:
si molta differenza no wobbling qualità superiore

zagarolo, 2017-11-17:
ci sono un po di peli

gregsaun, 2017-11-17:
This is due to the filament, you can remove them using a hair gun : 
https://www.youtube.com/watch?v=uqUcrIsSHak
You can also play with retraction but I never tried

gregsaun, 2017-11-17:
You can try to reduce temp a bit as well, I am at 225°C

zagarolo, 2017-11-17:
print 205

zagarolo, 2017-11-18:
Hi problem wobbling
[image]

gregsaun, 2017-11-19:
It does not look to be a wobbling issue to me.

* If the part is brittle, then you should increase temp and print slower
* Clogged nozzle?
* Does the temperature variate during the print?
* Filament is getting stock or tangled

zagarolo, 2017-11-21:
[image]

https://github.com/prusa3d/Prusa-Firmware/issues/75#issuecomment-311386115

Sebastianv650/Prusa-Firmware/tree/MK2_Community_Mod

Very very good

gregsaun, 2017-11-21:
So good, I have to update my firmware!

zagarolo, 2017-11-21:
https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.1.0-RC2

Read

zagarolo, 2017-11-21:
[image]

gregsaun, 2017-11-21:
Yep I read it and just finished to upgrade my firmware, let's go to calibrate the K value! Very nice new feature!

zagarolo, 2017-11-21:
Is very very faster print

zagarolo, 2017-11-25:
Ciao mi servirebbe se possibile un grande favore non sono capace a disegnare e creare un file stl per ABS  e Mi servirebbe urgentemente una modifica sul file originale Prusa mk2s asse X end e motor adatti per cuscinetti igus 15 mm ho un grande problema ti allego la foto grazie 
[image]

gregsaun, 2017-11-25:
Could you describe your problem?

zagarolo, 2017-11-25:
Guarda la foto il cuscinetto si muove scende nella parte interna

Look at the photo the bearing moves down to the inside

zagarolo, 2017-11-25:
create file in abs and petg

zagarolo, 2017-11-25:
I would be very grateful if you could create the abs file and pet parts original X end axis, and the engine part

gregsaun, 2017-11-26:
I do not thing it is a big deal if the bearing are going a bit inside. For now I do not have lots of time, I would like to finish the bear upgrade first to start a new one.

What you can simply do is this :
- Download MK2s scad files from here https://www.prusa3d.com/prusa-i3-printable-parts/
- Install OpenScad
- Open bearing.scad and reduce bearing_diameter value by 0.1 o 0.05mm (top of the file) and save
- Open x-end-idler.scad, then hit F6, then File -> Export -> STL
- Open x-end-motor.scad, then hit F6, then File -> Export -> STL

zagarolo, 2017-11-26:
New project ?

zagarolo, 2017-11-26:
Vorrei investire sto cercando di fare qualcosa di grosso cerco persone per investire denaro in nuovi progetti aprire una fabbrica io sono un costruttore di case ma ho sempre fatto elettronica vorrei investire miei soldi in qualcosa di grande

zagarolo, 2017-11-26:
Spero che il tuo nuovo progetto usare componenti di alta qualità e tecnica superiore per pezzi

zagarolo, 2017-11-26:
Quanti anni hai ?

gregsaun, 2017-11-26:
Could you contact via my website here : https://www.ekunn.com/ -> contact

zagarolo, 2017-12-02:
help I have a height problem of carriage Y extruder slams in the pot how can I change the end of the stroke find the new zero point? I modified bearings with double bearing under the carriage Y but increased height thanks

[2 more replies]

closed 2017-12-19 by gregsaun