gregsaun/prusa_i3_bear_upgrade #11: MK3 PSU dimensions

gregsaun, 2017-12-18:
Hi all,

I am working on v2.0 (check dev branch) and I need the dimensions of the MK3 PSU and PSU cover (assembled or not). The MK3 PSU cover looks a bit different than MK2s but dimensions seems to be very close.

Thanks for your help :)

gregsaun, 2018-02-22:
Here are the needed dimensions. If you cannot measure all of them it will still better to have one than none. Thanks in advance :+1:

[image]

libxmike, 2018-02-25:
D1 - 48.85mm bottom - 48.50mm top
C1 - 99.60mm
C2 - 16.80mm
C3 - 63.30mm 
B1 - 3.80mm (there where the screw ist)
B2 - there is no material on B2 side

gregsaun, 2018-02-25:
Awesome thank you very much!

DoItWithASmile, 2018-02-26:
A1	 33.25 mm
A2	 24.50 mm
A3	  2.60 mm
A4	  8.60 mm
A5	 17.00 mm
A6	. 4.60 mm
A7 	  2.90 mm
A8 	 17.00 mm
A9	  3.00 mm

B1	  2.70 mm
B2	  2.70 mm

C1	 99.60 mm
C2	mm
C3	 63.00 mm
C4	104.00 mm

D1	 48.60 mm

gregsaun, 2018-02-26:
Thank you very much! Quite different from libxmike, hard to say which one is good, or if there are two different PSU covers... I took some margin for the psu_lower_mount, so please give me some feedback how it goes. Thx in advance!

DoItWithASmile, 2018-02-26:
I wouldn't consider it to much of a difference. Some parts are hard to measure, too.

I would say my setup is +/- 0.5 mm accurate. just got a new caliper, but still... Especially the printed parts seems to have a bigger tolerance

gregsaun, 2018-02-26:
I know it is hard to measure. Both dimensions should fit in my psu_lower_mount