gregsaun/prusa_i3_bear_upgrade #12: Extrusion lengths

coreykovacs, 2018-02-11:
In this drawing 

https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/dev/full_upgrade/for_mk3/doc/vslots_length.png

It has the side uprights at 334 mm, but doesn't appear to take into account the fact the bottom side extrusions are laying flat in the V2 model. Shouldn't they then be 354mm ?

gregsaun, 2018-02-12:
You are totally right. BOM has the correct length. Thanks :)

gregsaun, 2018-02-13:
fixed

coreykovacs, 2018-02-13:
Sorry to be a nag. I see you corrected the measurements on the side but the measurements in the center of the drawing are still wrong and say the Z parts are 334 instead of 354. Looks awesome though and I can't wait to build one.

gregsaun, 2018-02-13:
Damned! Thanks for your patience :bowtie: . Those heights will be increase soon (another bug found during the first build).

I do not have a MK3 (waiting my 2.5 upgrade), so would be nice to have someone that would like to try the adventure!

coreykovacs, 2018-02-13:
it's no problem at all and I'm glad to be of help in any way.

gregsaun, 2018-02-15:
Thank you for your help. I have updated the files for Half upgrade and Full Upgrade MK2(s) and MK2.5. I still have to work on MK3 (Z will be just a bit higher).

Note that I also removed the unused PDF except for drilling the Half Upgrade frame (some builder like to print it and double check the holes positions).

commit 6a310182 (2018-02-13): Closes #12
commit dc8aeaff (2018-02-20): this closes #12
closed 2018-02-20 by gregsaun