gregsaun/prusa_i3_bear_upgrade #13: Slic3rPE PETG print settings

fishmsk, 2018-02-22:
Hi Grégoire

Could you please show your Slic3rPE settings for printing with PETG (post screenshots or config files) ? 
You have a document https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/master/manual/print_settings.md with some of the settings, but there are a lot of other, retraction parameters, cooling, speeds, etc.

I'm new to printing with PETG, just tried to print a rod_holder.stl with default settings (temp 230C/75C) - the result is fine but there is stringing in holes and "non-print" travels..

I'm new to GitHub too. So If it is not the right place to place a post - please feel free to delete it.

gregsaun, 2018-02-23:
That is the correct place, issues on github are made for bugs, questions, enhancements and so on.

First of all, the PETG I am using likes low temp, which is pretty rare for this type of filament. Usually PETG is between 220 and 240°C.

It do not make excessive stringing. I use the default slic3r retraction parameters and use a hot air gun to remove them after printing (when necessary). However, I have reduced a bit the speed because PETG likes to be printed slowly.

Check out this video, there are some information about stringing :
https://www.youtube.com/watch?v=8_adY2K-YIc

I know I do not completely answer your questions but hope I helped you a bit... Do not hesitate if you need more information ;)

gregsaun, 2018-02-23:
btw, version 2 will be released next week :)

fishmsk, 2018-02-23:
Great news !

jringv, 2018-02-27:
I was just about to order v-rails for the v1, what is changed in v2?? I'm waiting to place my order for the v2.
Great project gregsau!! Thanks for your work, looking forward to the v2.

gregsaun, 2018-02-27:
V2 will be out today :-)

gregsaun, 2018-02-27:
Done!
https://github.com/gregsaun/prusa_i3_bear_upgrade/

closed 2018-03-11 by gregsaun