gregsaun/prusa_i3_bear_upgrade #16: MK3 Full: Y-Mount different from MK2(.5s)

DoItWithASmile, 2018-02-28:
The y-motor mount of the MK3 is different to the MK2s in one important way:
the MK2s mount uses 3 screws to fix the motor, the MK3 mount only uses 2. Is this intentional?

MK2s:
[image]

MK3:
[image]

Since both mounts are for NEMA 17, the holes should be same.

BTW: The after market EINSY board can be bought with an endstop! so even though Prusa Research does not ship the MK3 with one, it would not be a problem to reuse the MK2s mount design with endstop.

gregsaun, 2018-02-28:
This is a bug, it is like my MK3 part is an older one, it should have 3 screws. BTW, original MK3 has only two

gregsaun, 2018-02-28:
I have exported the last version, that was an export issue. I keep this issue open until I have double check all exports. Thanks for reporting this :wink:

DoItWithASmile, 2018-02-28:
I know the original MK3 version only has 2. However, in the forum there were different beefed up designs:
- 3 screws to the backplate (included drilling holes... with the V-slot we can do this way easier)
- 3 screws to the stepper motor
- and even one where the stepper motor was held from both sides (that is what I printed: the y-mount from the STLs and a mirrored version for the other side of the motor. Quick and dirty. 😈)

gregsaun, 2018-02-28:
Yes you are right, I really do not understand the design of the original mount. Only two screws to mount the motor plus the arm are thinner than MK2s design...

I do not know the third design, do you have a link?

BTW, really sorry for this mistake, you will need to print one more, I hope you have enough filament...

DoItWithASmile, 2018-02-28:
I can confirm the newly exported STL looks like the assembly instructions.
Will print those with the next batch. New spool of filament arrived today. I am safe. 😃

Link to design no. 3: Prusa forum, ["Fine tune your MK3 Y axis"](https://shop.prusa3d.com/forum/original-prusa-i3-mk3-f30/fine-tune-your-mk3-y-axis-t13851-s20.html#p63825), rendering from RHD
[image]

DoItWithASmile, 2018-02-28:
I guess so. Couldn't find specs on the screws though. I guess I have to unscrew one to measure it...
Might check tomorrow since the printer is running.

closed 2018-03-01 by gregsaun

Panayiotis-git, 2018-03-01:
If you intent to remove the screws from the back of the motor, be careful not to remove the rotor from the motor. It could result in a dramatic loss of motor torque at some steppers.

DoItWithASmile, 2018-03-01:
Good point. I just removed one. It's a M3 x 30 mm.