gregsaun/prusa_i3_bear_upgrade #17: End caps on z-axis missing?

DoItWithASmile, 2018-02-28:
The extrusion of the z-axis is not covered by end caps.
[image]

Suggestion: add a pair of end caps
- BOM: Add 1x pair of end caps, add 4x set screws
- add instructions to step 12 (at least for MK3)

gregsaun, 2018-02-28:
Yes I did not design them, they have to be a bit different than the one with the bear. I was hesitating between: 

- 3d printed one with set screws mount
- 3d printed one with self tapping screws on top
- Openbuilds end caps http://openbuildspartstore.com/end-cap/

What do you think?

DoItWithASmile, 2018-02-28:
To stay true to the Bear Upgrade, here is my current thinking:

1. use 3D printed ones with set screws
2. why not ever so slightly modify the end caps we have:
[image]

This should fit here
[image]

gregsaun, 2018-02-28:
The problem are vibrations, this is why I have decided to use set screws and not only something that "clip" inside. Those "feet" for set screws are quite fragile and it is not really possible to make them stronger. So I like to keep the third "foot" in case one break or if you would to add one more screw.

I think I will purchase some self tapping screws for testing a first

DoItWithASmile, 2018-02-28:
In your experience did one ever break off due to vibrations?

Even a Bear Upgrade will be suspect to vibrations. That's for sure.
However, end caps are not load bearing. And wonder how and why they would break. 🤔

gregsaun, 2018-02-28:
I was not clear sorry, this is not a problem of breaking due to vibration, but the noise they can make if there are vibrating (because not correctly maintained).

Yes for sure Bear Upgrade will vibrate. Or it means you are not printing faster enough :grin:

closed 2018-03-25 by gregsaun