gregsaun/prusa_i3_bear_upgrade #20: Calibration point can't be reach

gregsaun, 2018-03-21:
Issue from adam666 :

> Hello Greg
> There is a slight issue which I am facing. My printer is complete but during calibration, it shows left front calibration point can't be reached. Although when I check it manually, the pinda probe is exactly in the center of it. Please advise

gregsaun, 2018-03-21:
Which version did you build Full or Half Upgrade?

Which calibration point can't be reach?

adam6666, 2018-03-21:
Full upgrade Mk2s, it shows "LEFT FRONT CALIBRATION POINT" can't be reachead

gregsaun, 2018-03-21:
So this seems to be the first point, did it scratch your bed?

If you confirm it is the first point, then try to lower a bit the PINDA probe.

adam6666, 2018-03-21:
Yes, it is the first point, but it didn't scratch the surface. I will lower probe a bit more and see what is the result.

gregsaun, 2018-03-25:
Hi adam6666, did you solve your issue? Can I close?

gregsaun, 2018-04-07:
I am closing it because no more activity

closed 2018-04-07 by gregsaun