gregsaun/prusa_i3_bear_upgrade #27: Can´t open files

straith2012, 2018-05-05:
Hello! Is there a way to open those files? I can not open any file

straith2012, 2018-05-05:
I can not open any stl. I've tried Cura, Slic3r and simplify with the same results. 
Maybe I'm doing something wrong, I don't know. I asked a friend and he had the same results
here are some images:
https://ibb.co/dF0Jon
https://ibb.co/jQbDM7
https://ibb.co/iYuNES

straith2012, 2018-05-05:
[image]
[image]
[image]

gregsaun, 2018-05-05:
wow that is a nice bug! I am looking at it now. thanks for sharing

gregsaun, 2018-05-05:
could you tell me how you get the files? Via GitHub, a clone or downloading the zip file? Or via thingiverse?

straith2012, 2018-05-05:
I got them through  Github. Now that you mention it, downloading everything using the zip seems to work. I feel so noob now but I guess downloading each file should have worked too. Thanks you very much

gregsaun, 2018-05-05:
No you are right it is not very clear. Yes download each file should work as well, and it is for me at least. Could you explain me how did you proceed?

straith2012, 2018-05-05:
Since there's not a download option for individual files, I just rightclicked the file and press "Download as". I made it that way because I already downloaded other files from github and never had an issue. I though something was bad because my friend had the same behavior, but I still don't know if it is an error or what

gregsaun, 2018-05-06:
Ok I see, yes I am not surprised it does not work. If you want to download each files separately you need to click to view the STL file rendering and then you can click on the Download button on the right. This not the best interface. If I have an idea I will do it, for now I do not want to generate a zip file each time I do a modification, this is not efficient. I am closing this issue then.