gregsaun/prusa_i3_bear_upgrade #30: (Full Upgrade) Manual tells to attach z-tops before inserting teenuts for PSU/Rambo mount

jhoff80, 2018-05-05:
It wasn't anything I had any problem figuring out, but towards the end of the build the manual gets a little bit rougher around the edges.  The biggest thing is that it tells you to attach the Z-tops which then prevents you from being able to insert teenuts for mounting the PSU & Rambo case.  Like I said, pretty minor as it's easy to figure out, but seemed worth noting.  (I did a MK2.5 full upgrade).

gregsaun, 2018-05-06:
Hi, I am sorry but I am not sure to understand. The Z tops are mounted on the "other side" of the v-slot than the PSU or RAMBo mounts. Where did you get blocked?

jhoff80, 2018-05-06:
Sorry about that, I finished my assembly on Thursday and was bringing this up from memory.  You're right, it wasn't the Z-tops.  It's the LCD support / end caps.  Instructions tell you to attach the end caps which block the slots, and then assemble the LCD support.

commit 5e7b1d8e (2018-05-16): switch order assembly. This closes #30
closed 2018-05-16 by gregsaun