gregsaun/prusa_i3_bear_upgrade #33: Y belt tensioner

gregsaun, 2018-05-19:
Add a belt tensioner for Y axis

gregsaun, 2019-03-13:
That is very interesting, thank you. I am still not sure I like the tensionning but I think I will give a try to your way.

wavexx, 2019-03-13:
After using this for a while, I can say I wished I had control over the tilt of the X idler as well. I can now track the belt perfectly on the Y to the point that I re-aligned the belt path with a calipher (moving motor and idler in line with the carriage), and *then* adjusted the axis until the belt tracks in the center.

Sadly I cannot get the same perfection on the X. Due to the way the nylock bolt holds the X idler, going off there is pretty easy even if the part is printed correctly. I'm trying to think of a way to allow at least one end of the idler to be aligned with a screw, but there's not a lot of space to play with :/

gregsaun, 2019-08-21:
This is now fixed with the new release https://github.com/gregsaun/prusa_i3_bear_upgrade/releases/tag/2.1.0-alpha0.

closed 2019-08-21 by gregsaun