gregsaun/prusa_i3_bear_upgrade #37: [Full MK3] Z Motor mounts are missing material

danclarke, 2018-06-24:
On the MK3 the standard mounting is with M3x10mm bolts rather than 12mm, and no washers. This means while the rear of the motor makes good contact with the mount, the front furthest from the aluminium extrusions is far from flush.

petethepig, 2018-06-26:
Yeah, I was also confused about that when I was looking through all the 3d printed parts.

There's this extra part called [z_motor_mount_front.stl](https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/master/full_upgrade/for_mk3/printed_parts/stl/z_motor_mount_front.stl). Here's [the page from the assembly manual](https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/master/full_upgrade/for_mk3/manual/assembly_instructions/step12.md). 

[image]

danclarke, 2018-06-26:
Oh that's excellent, glad it was my mistake. I think I'll make a small PR to amend the docs in a few places. They're already very good, but a slight clarification here and there I think will make them even better for complete novices like myself.

closed 2018-06-26 by danclarke