gregsaun/prusa_i3_bear_upgrade #39: Use of regular 2020 T-Slot extrusions possible?

Drachenkaetzchen, 2018-07-12:
I have quite many 2020 T-Slot extrusions left over from a previous build. From the documentation it is not clear to me if V-Slot is really required, or if I can use regular T-Slot extrusions?

I know that the build uses 4020 extrusions, but if V-Slot is not really required, I'd give it a shot with 2020 T-Slot extrusions and share my experience.

Drachenkaetzchen, 2018-07-12:
To answer my own question: The V-Slot is required because the 3D-printed parts are designed that way.

I will see if I can modify them to fit regular T-Slot extrusions.

gregsaun, 2018-07-12:
Hi Drachenkaetzchen , sorry for my late answer...

Yes parts are design for v-slot. However, if you open the source file on F360 you will find parameters to set width and height of the tabs. Angle cannot be changed for now.

Also check this remix here : https://www.thingiverse.com/thing:2872569

Here are some reason why I use v-slot (not ordered) :
- V shape gives the ability to make accurate vertical alignment tabs like on this part : [rod holder](https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/master/full_upgrade/for_mk2_mk2s/printed_parts/stl/rod_holder.stl). Thanks to this, almost no need to use a square to check alignment of printed parts. With t-slot you will need support and it comes with inaccuracy...
- Opensource with good community, FairShare program and regular contests with very good price
- Available in Europe and almost worldwide (Misumi is not available in Europe for private customers)! I know you can get T-slots from other places but they does not have the same profiles...
- 90° aluminum plates gives the perfect way to square the frame, even if cut is bad or dust is going in between. Plus 90° plates are very rigid! Check this drawing to see what is the effect of the thickness of a hair on a side of the extrusion : [effect of angled v-slot](https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/master/doc/effect_of_angled_cut.jpg)
- V-slot can be used as linear rails (will come on the Bear after my extruder rework)
- Openbuilds offers a complete set of hardware build around v-slot made for automated machines

Drachenkaetzchen, 2018-07-12:
No need to be sorry - I appreciate your feedback and it's not late at all! 

My motivation to use my existing 2020 extrusions is mainly because I'm on a budget, and I really want to get rid of the threaded M8 rods (like seen here: https://upload.wikimedia.org/wikipedia/commons/thumb/6/6a/Prusa_i3_with_metal_frame.jpg/800px-Prusa_i3_with_metal_frame.jpg ) - which is pretty awful for alignment anyways.

Also, from a quick look at #4 it seems there are no distributors in Germany where I could order them pre-cut. If I can relatively easily modify the Fusion360 sources to match my extrusions, I'll give it a shot.

Also, because my other 3D printer (MendelMax2 with heavy backlash issues) I couldn't guarantee that the printed parts are perfectly aligned, so I have to check everything manually as well.

Drachenkaetzchen, 2018-07-12:
Yes I've seen the store list, but well money is really an issue for me, so I try to use the existing parts which I have in stock. The link to the Thingiverse remix is pretty neat. You can close the issue if you wish. I might post it to this issue if I completed the build :)

gregsaun, 2018-07-12:
feedback is always welcome :-) 
All doc is here : https://github.com/gregsaun/prusa_i3_bear_upgrade/tree/master/full_upgrade/for_mk2_mk2s/manual

Happy building!

closed 2018-07-12 by gregsaun

fdjean, 2018-07-13:
@Drachenkaetzchen 
I bought my aluminum bars at [Modetis.com](https://www.motedis.com/shop/) in Germany (I-Typ Nut 5).
I have the STL for the half upgrade if you need

Drachenkaetzchen, 2018-07-13:
@fdjean that would be awesome! Do you have them available to download somewhere (ThingiVerse?)?

fdjean, 2018-07-13:
@Drachenkaetzchen In my fusion360 storage  ;) I'll publish on thingiverse when I'm back home

gregsaun, 2018-07-13:
@fdjean thank you for sharing it :-)