gregsaun/prusa_i3_bear_upgrade #40: MK3 Full Upgrade Step 6 Missing info?

ftKnox, 2018-07-13:
Is it just me or does step 6 in the full upgrade manual not show the required 4 tee nuts for the hinge and power supply mounts?

gregsaun, 2018-07-13:
You can insert them from top later :-)

ftKnox, 2018-07-14:
wow that's embarrassing... now to quickly close this before anybody see's it :)

closed 2018-07-14 by ftKnox

gregsaun, 2018-07-14:
hahaha you are not the first to ask for this! There is no stupid questions :-)

yellobello, 2019-05-29:
@ftKnox 👍 Wow, it´s closed for over a year now but still not in the manual ... just had to open the frame again to insert t(h)e nuts.

gregsaun, 2019-05-30:
@yellobello sorry for the inconvenience and time it takes. I was really busy with the extruder. Will be back on the frame soon.

In the mean time you can create pull request, that would be very helpful. If you do so please do it for the 3 versions: MK2s, MK2.5 and MK3.

yellobello, 2019-06-01:
@gregsaun To be perfectly embarrased, I too had to realize that most of the nuts can just be dropped in from above. But I will do so this weekend, so you just can OK the request. I am very happy that you are working on the extruder, by the way! so please continue ;-)

gregsaun, 2019-06-02:
Thanks a lot!