gregsaun/prusa_i3_bear_upgrade #53: Step 12.1 Could use a warning about Z motor cable lengths

bwcarty, 2018-09-11:
It has been a long time since my last Prusa build, and I forgot that the left z motor has a shorter cable than the right z motor. I ended up having them reversed until it was time to connect the cables, and I ended up having to disassemble a lot of parts to get the motors in the right place. 

A simple "Caution: Be sure the z motor with the shorter cable is on the left side. The z motor with the longer cable should be on the right" would save others the same grief. This could be added to step 12.1 and or fig 12.1.

iandouglas, 2020-04-12:
Would be good to clarify where the washers go around the Z motors as well.

gregsaun, 2020-04-29:
This will be addressed in 2.1 release.