gregsaun/prusa_i3_bear_upgrade #61: V-Slot parts and screws

DEvil0000, 2018-10-01:
With this build you need a lot of different none standard length v-slots. (You can buy them here in 5cm steps pre cut). It would be nice to have more of them the same length and in 5cm steps of length e.g. 350mm instead of 331mm.
For the screws its quite the same. Try to use the same legth everywhere and standard length like 10mm/16mm/20mm (8mm and 12mm are also fine but are normally not included in a set - here).
Additionally tee nuts are not a good choice.
It would be handy to have a step file combining all the parts as a overview.
I did quite the same as your full upgrade with my own design with standard v-slot length and only 20x20. Your design is better in a lot of ways but you need to cut the v-slots to quite some strange sizes.

matthew-humphrey, 2018-10-02:
I think the lengths were chosen so that the original Prusa parts would fit and provide an identical build volume. A big goal of this project is compatibility with the original (unmodified) printer.

gregsaun, 2018-10-03:
@DEvil0000 thanks for reporting this. I am interested to know where are you buying your v-slot? Do you have a link about your printer?

As Matt explained, one of the main goal is to propose a simple upgrade for Prusa printer, so I have to follow some dimensions. I agree with you that more "standard" lengths would be easier to manage for home made builds. I am planing to improve this on the next version, but I am not sure I can achieve it (specially on the Y axis). I prefer to have properly managed tolerances and strong printed part than standard vslot lengths.

8mm screws are what works best with Openbuilds plates so I will not change that one. 15mm will be updated to 16mm on the next version as it is not a standard length. I will have a look if I can switch the 12mm length to 16 or 10mmm, that is a good idea.

Could you tell me more about the tee nuts? What is wrong?

Good idea for the view of all parts!

DEvil0000, 2018-10-04:
The parts in this [repo](https://github.com/DEvil0000/Original-Prusa-i3/tree/MK3.5) are what I am using but maybe not 100% complete.
I am located in germany and was buying my v-slot at amazon (you find it when you search for e.g. ["Aluminium 20x20 150mm"](https://www.amazon.de/gp/product/B00K87LURW/ref=oh_aui_search_detailpage?ie=UTF8&psc=1) or "Aluminiumprofil 20x20 150mm"). There are at least 3 traders at amazon selling it: qfix robotics (where i got most of mine from), 3Diix and Dold Mechatronik (at their [items](https://www.amazon.de/Aluminiumprofil-20x20-I-Typ-Nut-50mm-2000mm/dp/B07D7BMG11/ref=sr_1_13?ie=UTF8&qid=1538637541&sr=8-13&keywords=Aluminiumprofil+20x20+150mm) site you can choose every 1cm). [qfix robotics](http://www.qfix.de/) also have a [webshop](http://www.qfixshop.de/) which is not working at the moment.
I know [Watterott](https://www.watterott.com/de/Mechanik/V-Slot-Profile) is also selling them which was the favourite shop at my university.
I had best results with those [nuts](https://i.ebayimg.com/images/g/VhoAAOSwMVdYGu91/s-l1600.jpg) - they give a strong and aligned hold.

gregsaun, 2018-10-04:
I agree that those nuts are better but I prefer to not have their extra tab in the middle and put this extra tab on plastic parts. This makes plastic parts stronger (and a low quality metal tee nuts will be still stronger than plastic).

Thanks for the store links, looks like you are using T-Slot mostly and not V-Slot (except if you are getting them from Watterott). I like vslot because you can print alignment tab vertically (you will need to use a support for t-slot and it is not reliable), plus it is open source with a useful community. In case it might be useful, have a look to the Openbuilds store list here : https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/master/doc/openbuilds_stores_list.md

If you like you can fork my project (it is under GPL because I have to use the same license than Prusa, so you will have to keep this license). However, I am doing everything under Fusion 360 so not sure you will like that...

DEvil0000, 2018-10-04:
> looks like you are using T-Slot mostly and not V-Slot

You are right I am using T-Slot. I was not aware of the difference till now.

> I am doing everything under Fusion 360 so not sure you will like that...

Not much since its quite expensive if you build something occasionally. I know its good and maybe worth it but I build something twice a year up to once a month. OpenScad works fine for me - maybe you can export the models as scad files?

gregsaun, 2018-10-04:
You should have a look to Openbuilds (inventors of the v-slot), they are doing very good job!
https://www.youtube.com/watch?v=584Z47gTBRY
https://openbuilds.com/

I found it much faster to use than Openscad but we all have our own habit. Unfortunately I cannot export a scad file (I even do not know a tool that can do that).

closed 2019-05-30 by gregsaun