gregsaun/prusa_i3_bear_upgrade #67: MK3 X-Axis issue

Pathogenius, 2018-10-28:
The last two twist ties on the extruder that hold the cable bundle to that rear arm needed to be removed for XYZ cal to complete, because they cause an end stop, removing them allows for an extra couple of millimeters to the left, thus the extruder then makes the end stop collision properly.

MK3 with R3 extruder

Pathogenius, 2018-10-28:
I neglected to mention probably the main contributing factor, that I am using the X-Axis parts from the 'bear_extruder_and_x_axis', but NOT using the extruder parts themselves.

Pathogenius, 2018-10-28:
I also neglected to mention that the collision is specifically into the RAMBO cover door, which does extend a fair amount in from the frame itself.

gregsaun, 2019-03-12:
Unfortunately the extruder and X axis are made to work together and are not compatible with other projects

closed 2019-03-12 by gregsaun

Lichtjaeger, 2019-03-12:
It would make me really happy if I can use a bear x-axis with a Prusa extruder someday.

gregsaun, 2019-03-13:
@Lichtjaeger To make this possible I had to redesign how the extruder is assembled in order to have room on the back for the new belt path. I also don't see in what the Prusa extruder could be better than the Bear extruder. What would be the motivation for that?

Lichtjaeger, 2019-03-13:
Sorry, I didn't mean to attack you. You've done an incredible job here.
It's just that, to my knowledge, there is currently no Bear MK3S extruder for MMU2S.

IMO it would be nice to be able to use Prusas latest innovations with your optimized designs. Interchangeable and modular as possible. So we can try both designs and combinations of them without unnecessary effort.

I only have one printer. When I want to try Prusas new extruder I can't replace the extruder only but have to swap the whole x-axis.

gregsaun, 2019-03-13:
Oh sorry for my message, at no moment I thought you were attacking me :-D . I was just asking awkwardly.

I am working on MK3S and should be out this week. MMU2S might take a bit of time but a member of the community is working on a solution right now : https://www.thingiverse.com/thing:3472911

[image]

[image]

[image]