gregsaun/prusa_i3_bear_upgrade #76: Slic3r-PE print settings

wavexx, 2019-02-25:
Some questions regarding the preferred print settings for recent versions of Slic3r-PE:

- Any explicit detail about why "thin walls" detection should be off? I also mostly print mechanical parts, and the external sizes of sloping/variable-width walls is overall greatly improved with "thin walls" enabled. I'm aware of a few corner cases though, so It'd be helpful to know if this was necessary or just "left as a default".

- "Ensure vertical thickness" can greatly affect the finish of some sloping parts, more than "think walls" by my experience since slic3r tends to create two distinct regions of rectilinear infill, one of which is very thin (causing vibration and overheating on the piece sometimes). With enough perimeters, I do tend to prefer to leave it off.

- Likewise, a small Infill => "Solid infill threshold area" (25mmq) is beneficial in my opinion to avoid small gaps just next to a solid area. I know attention was given to how pieces are filled, so this is something that might make a difference. Anything to watch out for? (regions that *should* be void -- for example)?

Thanks a lot :)

wavexx, 2019-03-04:
I went on and I printed all the parts. With 4 perimeters, "ensure vertical thickness" can be turned off for better results.

"Thin wall detection" does affect the clips and the rambo hinges (especially the lower one). For other parts there's substantially no difference, so the current reccomendation to leave it off is fine.

The 25mmq minimum solid infill should be on IMHO. This avoids generating infill for very thin gaps, where we extrude walls for <1mm with lots of travel (quite noticeable for gyroid). 25mmq is not enough to avoid this issue everywhere. But there's no clear cut... ideally this shouldn't be based on the area, but the average distance between the walls to fill.

gregsaun, 2019-03-12:
Thin wall may affect selective infill and might make some print looking very weird. To my experience this should be used only when really necessary.

For ensure vertical thickness I have to test this, thanks for the tip.

Small infill is interesting but I want to stay close to stock parameters. Thanks for this tip too.

closed 2021-11-27 by gregsaun