gregsaun/prusa_i3_bear_upgrade #79: MK3S: incompatible belt tensioner

wavexx, 2019-03-04:
Just a heads up:

The MK3S includes a belt tensioner just under the Y carriage, which is pretty nice given that there's no tensioner currently on the full upgrade (I had to play with the Y idler screws).

I thought I could retrofit the MK3S tensioner while doing the full upgrade, but it doesn't work ;)
The MK3 tensioner doesn't have a straight path between the idler and the motor anymore: the path from the idler to the carriage has been moved down due to the screw.

With the bear y idler, the lower path of the belt will touch the MK3 y tensioner when moving towards the LCD and pull the belt up as well. Since the entire tensioner is taller, a 16T idler won't work either.

However, I do still like the MK3S tensioner design :]

gregsaun, 2019-03-12:
Sorry for my late answer, I will create a new Y axis soon that will fix this issue :-)

lsarios9144, 2019-07-25:
Was This issue ever solved? Waiting to order the Bear Frame.

cggorman, 2019-07-25:
Not by Greg but the RHD tensioner is listed by him in the optional parts.  I can confirm it works on my Mk3S.  I'm actually using the toothed (20t, LDO) idler remix, tho.

gregsaun, 2019-07-25:
I am working on it right now ;-)

cggorman, 2019-07-25:
Greg,

Unrelated, but can you explain why the X & Y motor mounts only use three screws?

gregsaun, 2019-08-21:
@cggorman On Y you will hit the belt holder with 4 screws, at least on MK2S/2.5(S). On MK3 it would be too small to be efficient.

gregsaun, 2019-08-21:
This is now fixed with the new release https://github.com/gregsaun/prusa_i3_bear_upgrade/releases/tag/2.1.0-alpha0. Thanks for having reported this!

closed 2019-08-21 by gregsaun