gregsaun/prusa_i3_bear_upgrade #82: It's Bear Upgrade ready for MK3S?

Sugar250, 2019-04-16:
Hi,
I just want to build a fresh new Prusa i3 MK3S by my own, and I want to use bear upgrade.
I fond that in issue #67 you said you are working on MK3S and should be out that week. So is it right to be use now?
Thanks for your incredible job.

Sugar250, 2019-04-17:
And also #79

rafaljot, 2019-04-18:
I upgraded my mk3s using
- https://www.thingiverse.com/thing:3469271
- https://github.com/gregsaun/bear_extruder_and_x_axis/tree/dev/printed_parts/stl/mk3_mk3s
- https://github.com/gregsaun/bear_extruder_and_x_axis/tree/dev/printed_parts/stl/common_to_all_versions

gregsaun, 2019-05-30:
Yes but you have to use the belt holder from MK3. Y axis will get a rework to add tensioner on all versions

closed 2019-05-30 by gregsaun

codingcatgirl, 2020-07-26:
To anyone who might be reading this (outdated) issue looking for an answer, let me save you some time: The dev branch is compatible with the MK3S, and any 2.1 release is as well, you can just check out that tag.