gregsaun/prusa_i3_bear_upgrade #83: New Delta PSU

Wiggleplum, 2019-04-22:
It appears Prusa is shipping a new PSU made by Delta with their kits, and is sending them out as warranty replacements. It seems that the lower PSU mount can not be used with it. Are there any brackets available for the full bear? I received mine today, and am wondering the best way to support it besides the two screws on the side. 

https://manual.prusa3d.com/Guide/7.+Heatbed++&+PSU+assembly+(Black+PSU)/1153

matthew-humphrey, 2019-04-22:
The lower PSU mount is completely unnecessary. The upper mounts are more than sufficient to suspend the PSU.

Wiggleplum, 2019-04-22:
> 
> 
> The lower PSU mount is completely unnecessary. The upper mounts are more than sufficient to suspend the PSU.

Good to know. Thank you.

gregsaun, 2019-09-15:
Fixed with latest 2.1.0-alpha1 release.

closed 2019-09-15 by gregsaun