gregsaun/prusa_i3_bear_upgrade #84: File Error in Slic3r PE

TimberRocket, 2019-05-02:
Slic3r PE isn't able to open any file of the full bear upgrade for the Prusa i3 Mk3.
Everyone shows an error and just closes. Is it a known issue?

gregsaun, 2019-05-02:
Could you go on the main page:
https://github.com/gregsaun/prusa_i3_bear_upgrade

Click "Clone or download" and then "Download ZIP" and try again?

TimberRocket, 2019-05-03:
Thank you for you fast respond :)
I'll try it as soon as i'm home.

cggorman, 2019-05-03:
Are the v-slot frame parts the same for mk3 and mk3s?

stahlfabrik, 2019-05-03:
Yes

cggorman, 2019-05-03:
Thanks!  I'm brand new to github and 3d printing.  Just ordered a mk3s and trying to learn everything possible. Planning a full Bear as soon as I get the printer up and running properly.

TimberRocket, 2019-05-03:
Gregsaun...
Tried downloading all files as copy (master) from the index page. Works great!
Seems that the files are somehow damaged while downloading them part by part

closed 2019-05-03 by TimberRocket