gregsaun/prusa_i3_bear_upgrade #95: Step 9: MK2/2.5 Upgrade Using MK3 Frame

Menithal, 2019-10-07:
Step 9 needs to be updated with the MK3 frame dimensions:

Y motor mount on MK2/2.5 is different from MK3 Y motor mount, so cannot use dimensions from the MK3 if you have MK2/2.5 Mount printed.

Alternatively readme warning should also include to print Y motor mount from MK3 instead of using the one from MK2/2.5, similar to using the alternative Rod Mounts to match dimensions.

gregsaun, 2019-10-07:
You have to use the MK2/2.5 Y motor mount because you need the end stop switch. So you have to follow step 9 of MK2/2.5 assembly instructions.

Menithal, 2019-10-07:
OK yeah, I had some confusion there so my measurements were still using MK3 measurements (rod distances), gonna adjust them to match MK2.5, some clarity could be put into the instructions though what parts are different on the MK3 frame.

closed 2019-10-07 by Menithal

gregsaun, 2019-10-07:
No problem, happy we solved it. I am working on new version with new instructions, they will be much much better ;-)

Check out the work in progress of the future version, there is a new Y axis with tensioner now:
https://github.com/gregsaun/prusa_i3_bear_upgrade/tree/dev