gregsaun/prusa_i3_bear_upgrade #98: Layer shift issue

gregsaun, 2019-11-09:
I looked at the 2.1 Alfa and did download the mount. Am waiting for the final of 2.1 before I commit to final assembly. Also. after 3 successful prints last night, after replacing all Y axis rods and bearings, my print tonight had a 1-2 mm shift at about 20mm height (z). I guess that was not the problem. Plenty of grease and no marks on rods. The problem is elsewhere. Any thoughts???

_Originally posted by @taz43mca in https://github.com/gregsaun/prusa_i3_bear_upgrade/issues/96#issuecomment-552055884_

gregsaun, 2019-11-09:
Hi @taz43mca, I have created a separate issue to better track your problem.

First of all, are you sure the shift is in Y and not X (or both)?

taz43mca, 2019-11-09:
Just in y axis. Also, just did upgrade few to 2.8.1 one other person had similar problem after upgrade. He went back to 2.8.0 and problem went away.

gregsaun, 2019-11-09:
Do you use bearings from Prusa? Are you using the U bolt or the Bear bearing holders?

taz43mca, 2019-11-10:
The bearings are the original ones from PRUSA. The replacement one are off the shelf as I did not see any ones on the PRUSA web site.I used the original "U" bolts as this is an unmodified (BEAR) printer.After taking the advise of one of the posters and flashed the printer with the older 3.8.0 Firm Ware, the problem went away. Changing the rods and bearings were not a waste but seems something is wrong with FW 3.8.1.Note the BEAR bearing holders will not clear the PRUSA original frame. Tried that. Have now had the printer running for a 15H 30 M print and now a 7 Hour print and NO PROBLEM or SHIFT. Used both PLA and PETG and Sliced with the latest Slicer update.Hope this helps.Note, I am building a MK3  BEAR from the ground up. Like the BEAR Frame, Updated Extruder and x axis belt tensioner, and other improvements.

gregsaun, 2019-11-11:
Thank you for those information. On the Prusa's github there is an issue talking layer shift after a power panic occured. Did you had a power outage recently?

In the meantime it could be great to double check the distance of your smooth rods. Use a caliper and measure directly from rod to rod directly. If your caliper is too short, you can sometimes remove the little end stop at the end of your caliper and have a bit more length (generally enough to measure the inner distance).

taz43mca, 2019-11-12:
No power loss/outage or panicThought of it and did measure space between rods, all checks out. Did that when I replaced rods and bearings, also checked belt tension, greased the bearingsflashed 3.8.0 firmware (had done recent flash of 3.8.1 before problems started)then I printed 3 objects: 15+ hours, 6+ hours and 3+ hours with no shift and all prints as good as with new printer.Two of these were printed from the same files that had been a problem before. One was a new slice.Note: I have 3D printed a spacer that makes sure rods are spaced exactly as required.
Has anyone checked FW update 3.8.1 on a MK25 to see if it gives  print shifts. They may have to print a tall and large object to check. As of right now I will continue using FW update 3.8.0 as it works.

gregsaun, 2019-11-12:
I could not find someone reporting that on the Prusa's github, maybe it could be great to report one then.

Indulae, 2019-11-16:
I have the same issue of layer shifting only on y axis.

taz43mca, 2019-11-16:
are you using a Prusa i3 2.5 or 3 and FirmWare update 3.8.1?I went back to FW update 3.8.0 and problem went away.

taz43mca, 2019-12-01:
In discussing this issue on the PRUSAPRINTER.com blof/forum, one person noted that he has reflashed his printer with the 3.8.0 firmware and that solved the problem. I had done the same thing and the problem went away. I have just done two prints, one at 197mm that at the 194mm level has what appears to be incomplete OUTside layer. The infill appears to be there but the outside smooth layer does not. The other print as 134mm high and had a similar problem at the 126mm layer. I can take pictures of both (note that this is when I am using the 3.8.0 firmware. I have updated my slicer and drivers to the latest posted on the PRUSA web site. I have a WIN 10 pro OS.
Your thoughts?Merrill Albury

gregsaun, 2019-12-05:
Sorry for my late answer. Could you share a picture of your printer? Sorry to ask for that but most of the time a picture/video saves tons of word.

In the meantime, check that your X axis and your filament aren't catch by sometimes when you reach top of Z axis.

closed 2020-03-10 by gregsaun