gregsaun/prusa_i3_bear_upgrade #99: MK3 Full Frame - MK2 parts using Revision 2.1

aceycv8, 2019-11-19:
Hi - i think i know my own answer but want to double check before starting.

I've ordered a full MK3 frame and preparing all the STL's to print.
Which instructions do i use to build the frame from scratch? 
Do i use the MK3 instructions but use the "mk2s_on_mk3_parts_positions" to mount my parts?

Thank you

gregsaun, 2019-11-19:
Hello, thank you for your order, hope you will like it :-)

You have to follow the instructions from the temporary assembly guide for all printed parts (parts position are describe in there). For the rest follow the instructions from Bear 2.0 Full Upgrade MK3 assembly (like for extrusions).

gregsaun, 2019-11-19:
Bear 2.1 manual is here:
https://github.com/gregsaun/prusa_i3_bear_upgrade/tree/dev/manual

Bear 2.0 Full Upgrade MK3 is here:
https://github.com/gregsaun/prusa_i3_bear_upgrade/tree/master/full_upgrade/for_mk3/manual

aceycv8, 2019-11-20:
Thankyou gregsaun - can't wait to see it in action!

closed 2019-12-13 by gregsaun