gregsaun/prusa_i3_bear_upgrade #100: Y Motor Mount Lower Extrusion screw too short

ctilley83, 2019-12-13:
The new revised Y motor (2.1) mount no longer utilizes the same length screw for the bottom screw hole.  For those with version 2.0 frames, this conversion requires a screw that is 2mm longer than stock.  This could be remedied by deepening the pocket for the lower horizontally facing extrusion mounting hole so the existing screw will properly engage the t nut.

gregsaun, 2019-12-13:
I tried to do that but it isn't possible. The minimum acceptable would be M5x15mm screw to not hit the M3 screw. However, 15mm screws is not a standard length (even if sold by Openbuilds) and to give us the choice to use better screws and user to source easily the parts we went to M5x16mm screw.

If you make it shorter you will have a hole in the M3 screw path which can affect the tightening of the stepper (lots of vibrations at this location).

You can see all this in the drawing bellow. The 11mm length in bottom is the M5x16mm length (5mm goes inside the t-nut). The 5mm on the left is half of the space needed for the M5 screw head.
[image]

ctilley83, 2019-12-13:
Gotcha. I see now.

closed 2019-12-14 by gregsaun