gregsaun/prusa_i3_bear_upgrade #101: New 2.1 assembly manual

taz43mca, 2019-12-14:
Any idea when the new 2.1 assembly guide will be finished. I have downloaded the temporary guide but am waiting for the finished product since I see discussions about it are still in progress. 
I had made spacers for the 2.0 version (MK 3) to ensure that all parts are properly placed and have also done similar spacers for the 2.1 (so far mods). As a note, I had made similar spacers for the old MK2/2.5 but since the kits are all MK3 did not want to post and confuse the issues. Have similar reservations about those made for the 2.0 since the 2.1 parts are different and require some spacers to be modified.
Thanks for all who are working to make a better printer. Hard to do since we daily see new and "better" ways to do things.

gregsaun, 2019-12-16:
I am working on the RC1 now and in parallel to the new hardware with my manufacturer. Once RC1 will be published I will start working on the assembly guide.

RC1 will include new build helpers for Y and Z axis. I push them to the dev branch last week:
https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/dev/printed_parts/common_to_all_versions/stl/build_helper_y.stl
https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/dev/printed_parts/common_to_all_versions/stl/build_helper_z.stl

The Y axis helper is going to be used like this:
[image]
Be careful to use it only on the left side and not both sides! It is not made to set the width of the smooth rods.

Do you have some images about those spacers, I am interested to see what you made?

taz43mca, 2019-12-17:
I have numerous photos. I will see if I can zip them small enough to send. Although I have been designing and producing spacers for all the variations, I will just zip those for the latest (MK3 frame and 2.1) changes.
I have not posted them to thingiverse nor have any experience with that site but planned to do so in the near future. Any suggestions?Carl

https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/dev/printed_parts/common_to_all_versions/stl/build_helper_y.stl
https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/dev/printed_parts/common_to_all_versions/stl/build_helper_z.stl

gregsaun, 2019-12-17:
The problem with printed spacers/helpers is that they can be inaccurate. Some people have badly calibrated printer and dimensions or accuracy can be wrong. This is why, for example, I recommend a quality square to build the frame and do not provide printed part.

taz43mca, 2019-12-17:
Mine measured just fine but understand your argument. Thanks for the feedback.

stelgenhof, 2020-01-18:
@gregsaun Looking forward to the new manual. I'm currently working on the v2.1 upgrade for my MK2.5 and a single overall manual would be very welcome :) Not a major burden, but now I have to navigate between the two branches to see if I follow everything in the right order.

gregsaun, 2020-01-19:
@stelgenhof The new manual is going to be like the Bear extruder. Keep in mind that 2.1 is still under development so for now the manual is just here to help assembly but will removed later.

taz43mca, 2020-01-19:
I am building a Prusa Mk3/Bear hybrid from scratch. Jut finished the Bear X axis extruder upgrade. I agree, a new 2.1 manual would sure make things easier. I have also noted that some of the items, like the Z axis motor mount shown in some drawings are different from the original Bear upgrade so I designed and printed them on my own. Makes the task of getting the motors level and at the right height much easier. I have done some more tweeking with other parts and designing and printing some spacers that make assembly much easier. As they say, the proof is in the pudding. Will post when I finish.

taz43mca, 2020-02-18:
Could not find a way to ask this but:
Build a Prusa MK3S Bear. Downloaded latest firmware (FW) 3.8.1 and flashed Rambo board. (did twice). Started self test. completed all motor and temp tests, when it got to the XYZ test/cal it tested movement of X and Y axis and the Z axis. All worked ok until nozzle moved to bed to do the 4 point leveling check. As soon as nozzle reached the bed, it stopped and I got a calibration error. Unit did NOT do any circular search as my MK2.5 does. It just reached the bed and stopped (not touching). I did a complete check of all my measurements and as per error procedures and Bear assembly manual, and all are OK. Did a reflash of the FW. Same results.
The Rambo board was acquired from ALL3D (dropped shipped). I could maybe understand if the printer ran a circular search before it gave the error but did not. I DID NOT press the reset button on Rambo board but thought I might do that and then reflash again. HELP!
Suggestions???

taz43mca, 2020-02-26:
I built a Prusa MK3S with the 2.1 mod., including the Bear extruder upgrade and I have a problem. All tests are OK until it starts the 1-4 bed Z axis cal. When the extruder comes down to the bed it stops and says calibration failure. I have reset and reflashed the Einsy board with various versions of the MK3S software and no luch. The PINDA tests fine and all XYZ axis checks say correct. If I do an auto home cal it does the cal over the target. When I do the XYZ cal that starts with and auto home it appears to do the test OK, raises the Z axis to the top, tells me to clean the nozzle, put a sheet of paper on the, no metal sheet. When I press the nob to start the test the extruder comes down and stops withing a mm of the bed, with the PINDA about 5mm to the left of the edge of the target circle
20200225_175327
and says calibration failed...refer to manual.
I have rechecked all my measurements and all are correct. What I did notice is that when the extruder goes all the way to the left and stops at the Z axis frame, the nozzle is about 3-4mm short of the 250mm line on the bed. When the nozzle is moved all the way to the right, it is 3-4mm past the edge of the bed.
Thought of moving the bed 3-5 mm to the right so that the nozzle is in line with the left dotted 250mm line when up against the stops but then the PINDA would be even further to the left of the target. The bed rails would no longer be 82mm from either side as shown in the drawings.
I have been fighting this for days and working with Jason at Ultimachine trying to resolve.
Any suggestions???

taz43mca, 2020-02-26:
[image]
   Here is a photo of where the PINDA sets

taz43mca, 2020-02-27:
With more extensive checking, the offset is 2.3 to 3 mm to the left

stelgenhof, 2020-08-24:
Awesome! @gregsaun Thanks for all the hard work. My v2.1 was just completed this weekend, with a first successful test print :)

gregsaun, 2020-08-25:
@stelgenhof Happy to read that, enjoy!