gregsaun/prusa_i3_bear_upgrade #105: Mk3 calibration failure with Bear 2.1 build.

taz43mca, 2020-02-28:
Unfortunately I posted my last on this on the 2.1 manual discussion. But here goes again.
Built a Prusa Mk3S from scratch using the Bear upgrade with the 2.1 mod. Also the bear extruder mod.
The kit came from a listed source including the Einsy Rambo which was dropped shipped with the Prusa software installed. Other PRUSA parts as needed.  Assembly went OK and all measurements checked OK.
Per Prusa instructions I downloaded the latest FW and SW from PRUSA web site (FW 3.8.1).
Started the full calibration and tests. XYZ motors, bed and extruder heater went OK. Got to the Z axis 4 point cal and after it went all the way to the top, had me clean the brand new nozzle and put a sheet of paper on the bed, tell them that there was no steel plate on the bed, it moved down to the bed and IMMEDIATELY said calibration failure. I have downloaded and reloaded the firmware (after doing the hard reset), even older versions 3.6.1; rechecked all my measurements; did a complete PINDA check and still get the same results. I tried an Autohome test and it did that right over the target but when it does one before the Z cal, it is over the side of the bed (the PINDA is about 2mm to the left and 2mm to the front of the target.
Then I noticed that when the extruder was all the way to the left it was 2mm to the right of the left print edge line (extruder is against the stops) and when it fully to the right, it is about 2mm outside the right edge of the board. The Y axis rods are exactly 82mm from the ends of the V slot per instuctions. I have been discussing this with Ultimachine (Jason) and he has given me many things to check but the results are the same. It is interesting to note that when I do the PRUSA hard reset per their instructions, it says it is fully reset but as soon as the printer starts up I get a "choose a language" and then it goes right to the PRUSA calibration menu. Seems that the hard reset does not fully clear the EEPROM of all PRUSA information. Maybe that is as it should be but if I bypass the test and it goes to the regular print screen it looks like the FW is still on the EEPROM. I thought about doing a couple of things:
1. press the reset button that is actually on the Einsy Rambo board - cannot find exactly what that does.
2. move the bed rails 2mm to the right so that, looking from the front, the left is 84mm from the end and the right is 80mm from the end. I would also have to move the Y axis motor and the idler 2mm to the right. This would, however, move the PINDA ferther to the right of the target when the extruder in the fuly right position but would put the nozzle directly over the left print line.
I have attached a PDF drawing of what I have right now.
HELP!
[Bear Drawing problem.pdf](https://github.com/gregsaun/prusa_i3_bear_upgrade/files/4269488/Bear.Drawing.problem.pdf)

gregsaun, 2020-03-10:
I have just realize I missed that issue, sorry for my late answer!

With the Bear extruder you have to use the Bear Calibration firmware as explained in the manual here:
https://guides.bear-lab.com/Guide/5.+Final+adjustments+and+calibration/14?lang=en#s244

This should fix your issue.

taz43mca, 2020-03-11:
In talking to Jason Reynolds at Ultimachine, he asked me to try the MK3 FW vice the MK3S. The MK3 did a cal with no problem except it would not recognize the "S" upgraded filament sensor. I thne reflashed with the <K3S FW and it failed as before. I assume that I need to do a factory reset of the printer and then reflash with the Bear FW, then reflash with the MK3S FW (no reset). BUT it would appear that I would need to use the BEAR FW if I found a need to recalibrate the printer for any reason. True?Regards,Merrill

https://guides.bear-lab.com/Guide/5.+Final+adjustments+and+calibration/14?lang=en#s244

gregsaun, 2020-03-11:
Yes you need to use the Bear calibration firmware as explained in our assembly guide. Prusa has changed the geometry of the extruder between MK3 and MK3S. Z height is a little different and so calibration fails with Bear extruder which uses a similar geometry than MK3. The Bear extruder didn't follow the MK3S design because we found that it can create several potential print quality issues as well as lowering the reliability.

taz43mca, 2020-03-11:
If I read this correctly, are you saying that I am better using the Bear FW for the future as it corrects the Prusa FW problems? If that is true then future Prusa MK3S FW upgrades would have to be complimented by a like Bear FW update. And does the difference in the FW still give you the extra 10mm (210 vs 200) not had with the MK2/2.5? Not that this is a game breaker.Regards,MerrillPS: I will give you feedback on how the calibration and subsequent print goes with the Bear FW.

taz43mca, 2020-03-12:
Update:I have not put it through all its paces yet but did a small test print that is acceptable. More Z cal may be needed to tune it.So what happened: I followed the instructions exactly but since I did a new build and not a modified MK3, the printer did not recognize the filament sensor and the Z axis test failed. So I went back and used the Wizard that tests everything and that did the trick. So a new build has to do things differently then a modified build.I plan to do some of my more intricate builds with color changes and see how it is doing...using the BEAR FW. Then I MAY load the Prusa FW and see how that does. When I loaded the file to print the printer told me to update the FW to 3.8.1 but I just clicked through it and when ahead and did the print.Why, you may ask, well I had some interesting problems with 3.8.1 FW and my MK2.5 (not 2.5S) and noticed that some others also had the same problem with layer shift. So my MK2.5 is running the 3.8.0 FW with no problem.After I put the Bear MK3S clone through its paces and after I use it with the Prusa 3.8.1 FW, I will give you a final update.For now, thanks,Merrill

taz43mca, 2020-03-13:
A follow up to my printing and using the Bear FW.I have a problem and don't know why. Using the Bear FW and the Prusa latest FW for the MK3S I have the same problem. When it starts to print after doing the 9 point check, it goes to the side and says "unload filament" I do that and then "load filament" and it returns to the bed and immediately does it again. It keeps repeating this. I tried using the Prusa slicer on two different computer to slice the print and write the g code; I used both an SD card and Pointerface with a USB cable to print the item, and still got the same results. So I disabled the filament sensor and got it to print. Any ideas why this is happening??? I thought all my troubles were over with flashing and doing a claibration with the Bear FW. This is new. Interesting that I was able to do a print without this problem after the initial wizard calibration. Nothing changed except I turned off the printer and turned it back on. I only made changes AFTER the problem appeared.Regards,Merrill

gregsaun, 2020-03-14:
Happy to hear that you solved the issue. This is the first time I heard about new printer requiring the wizard, I will keep an eye on this.

The Z height of MK3(S) is a long story. MK3 was announced at 210mm for Z (MK2S was at 200mm), but mechanically there were less than 5mm of difference on the frame and around 1mm on the extruder, not enough to reach 210mm. I suppose they wanted to fix this on MK3S so they did it by moving the extruder roughly 6mm higher up. The Bear extruder has not following this MK3S strategy because it can be an issue with temperature around the heatbreak (which was already an issue for MK3). The Z calibration algo and self test are testing this Z height and so it will fail with your the bear extruder.

taz43mca, 2020-03-14:
I guess you have not seen my note that with the successful calibration, my problems were over, was a bit premature. Seems that now, every time I try to print, the printer does it 9 point check, goes to home and then over to right edge and says "remove filament" after I do that, it says to put in filament and asks if the color is ok. Then it goes back to the home spot and back to the "change filament" again. It keeps repeating that and will not go any further. So:I tried slicing with the Prusa slicer on two different computers, thinking that I may have a corrupt G code somehow; then I tried printing with bot  an SD card and then Pronterface and a USB cable. Same result. Then I Flashed the printer with the Prusa 3.8.1 FW (up to this point I was using the Bear FW) and same result. Note that the first small print I did after a successful configuration, was done with a USB cable from my computer using Pronterface. The only thing I did between that small successful print was shut down the printer and later restart it. Note also that the object I tried to print was a new one and also used some older ones that had printed successfully on my MK2.5.Soooooo. I disabled the printer filament sensor and it printed OK. (turned off both Autoload and Sensor). Thought I may try with just one then the other enabled and see if that makes a difference. Interesting that my disabling the autoload did not disable the sensor but that may be the way the software works.Any ideas?
Regards,MerrillPS: Having to use the Wizard first may be due to the fact that this is a completely new Bear build with the Enisy Rambo board bought from ALL3D which came with the Prusa FW already flashed. I assumed that the procedure on the BEAR was for someone who had used their MK3/3S and now had done the Bear mod. That is why, when the procedure alone did not solve the calibration failures, I used the whole Wizard rather than the XYZ calibration alone.

gregsaun, 2020-03-14:
You are probably right for the wizard!

You definitely have an issue with the filament sensor. Make sure the little arm printed well. You can also try to test the sensor by placing something in the sensor (you can read the sensor status in the Support of the LCD menu). Make sure the arm moves when you add and remove a filament. If you still have an issue please share a picture here.

taz43mca, 2020-03-15:
will try.thanks

taz43mca, 2020-03-15:
Checked the movement of the arm and it is ok. The autoload works like it should. I did a printout of the screen info on Prontoface when it failed. Will check the LCD and get a video.On Mar 14, 2020 9:43 PM, imagesbyalbury@aol.com wrote:

will try.
thanks

taz43mca, 2020-03-16:
An interum reply.The sensor checked and works as designed physically. The autoload feature works.
I used Pronterface to print another item. At the time the Support-Sensor state menue was open:It read as follows:PINDA:   0FINDA:   N/AIR:          0
I started the print and here is what happened:9 point check went OK as usual; the extruder laid down the initial strip and commenced printing. The screen read as follows:PINDA:   1FINDA:   N/AIR:          0It continued printing until I pressed the knob to return to the main screen. As soon as I pressed the knob, the printer stopped printing and went to the right bed corner and did a filament unload.I unloaded and reloaded the filament; reset the printer, and started to print again but it did the 9 point check, moved to home, and then moved to the filament unload position.Another reset, with the support-sensor state menu open, and the print worked as it should. Again, as soon as I pressed the knob it stopped printing and moved to the filament unload position and said unload...After 3 or 4 more tries I successfully plugged the extruder so now will have to do some work to undo the jam.More testing to follow. BUT the big question is why would the position of the menu and pressing the knob affect how the printer worked?I will clear the jam, do another test, both ways, and make a short video I hope I can upload to you if you desire. All you will see is just what I described. I also thought about doing a copy of the TEXT on the Prontoface screen as the printer prints and fails. Would that help?Regards,Merrill

taz43mca, 2020-03-19:
Did a 4hour and 58 minute print with the menu set at "Support...Sensor state" with no problem. I plan to disconnect the Filament sensor and see if that allows me to print in a normal menu setting, not having to put in the "support..sensor state" position. I will advise.regards,Merrill
In a message dated 3/15/2020 8:59:28 PM Eastern Standard Time, imagesbyalbury@aol.com writes:

An interum reply.The sensor checked and works as designed physically. The autoload feature works.
I used Pronterface to print another item. At the time the Support-Sensor state menue was open:It read as follows:PINDA:   0FINDA:   N/AIR:          0
I started the print and here is what happened:9 point check went OK as usual; the extruder laid down the initial strip and commenced printing. The screen read as follows:PINDA:   1FINDA:   N/AIR:          0It continued printing until I pressed the knob to return to the main screen. As soon as I pressed the knob, the printer stopped printing and went to the right bed corner and did a filament unload.I unloaded and reloaded the filament; reset the printer, and started to print again but it did the 9 point check, moved to home, and then moved to the filament unload position.Another reset, with the support-sensor state menu open, and the print worked as it should. Again, as soon as I pressed the knob it stopped printing and moved to the filament unload position and said unload...After 3 or 4 more tries I successfully plugged the extruder so now will have to do some work to undo the jam.More testing to follow. BUT the big question is why would the position of the menu and pressing the knob affect how the printer worked?I will clear the jam, do another test, both ways, and make a short video I hope I can upload to you if you desire. All you will see is just what I described. I also thought about doing a copy of the TEXT on the Prontoface screen as the printer prints and fails. Would that help?Regards,Merrill

taz43mca, 2020-03-20:
After the 4 h 53 m print, I disconnected the filament sensor and tried to print with the menu in the normal position (not in Support...Sensor State selection) and it failed as before. I think that eliminates the filament sensor as the problem.Since it prints OK when the menu selection is "Support...Sensor State" , it still leaves the firmware (which has been flashed lastly to PRUSA MK3S ver 3.8.1. Same results as when I tried to print in the BEAR FW so think that may eliminate the firmware. That leaves the Ensiy Rambo and the LCD panel. Both of which I find hard to believe is the cause of the problem but seems there is no other choices.Thoughts.Regards,Merrill
In a message dated 3/16/2020 3:02:15 PM Eastern Standard Time, carlalbury@hotmail.com writes:

The video is well over the limit for sending. But here is what I tried:

Sliced as         SD Card to print                    Menu at Support; Sensor State      When pressing knob

MK3S              Failed after 9 point check    Printed OK                 until                Failed

MK3                Failed after 9 point check    Printed OL                  until                Failed  (note when loading file the printer said this was not correct printer..do I want to continue)

MK3S              Failed after 9 point check    Printed OK                 until it stopped to change color then failed after filament change (had to press knob to make the change so got

 out of “support: sensor state” menu selection.

(Note: I set this up for a color manual change at .40mm as set up in Prusa slicer)

 

Could this be an LCD panel or (heaven forbid) an Enisy Rambo problem? Or maybe something in the software although I have had the same problem with the Bear FW and Prusa FW.

 

 

Sent from  Mail for Windows 10

 

From: imagesbyalbury@aol.com
Sent: Tuesday, March 10, 2020 11:17 PM
To: carlalbury@hotmail.com
Subject: Fwd: [gregsaun/prusa_i3_bear_upgrade] Mk3 calibration failure with Bear 2.1 build. (#105)

Reply-to: reply@reply.github.com
To: prusa_i3_bear_upgrade@noreply.github.com
Cc: imagesbyalbury@aol.com, author@noreply.github.com
Sent: 3/10/2020 1:05:17 PM Eastern Standard Time

https://guides.bear-lab.com/Guide/5.+Final+adjustments+and+calibration/14?lang=en#s244

taz43mca, 2020-03-23:
Just got a post on the forum that said they had same problem with the printer going to "filament remove" position. They did a complete reset of printer and problem cleared. If I do that then I guess I will lose all calibration data and will have to flash with the Bear FW, recalibrate printer, then reflash with Prusa FW. Or is there a way around it.

taz43mca, 2020-03-28:
Tried but no help.

taz43mca, 2020-03-30:
Success. The IR lever was not tripping the IR sensor, even when it and IR sensor were physically removed from the extruder body and manually moved together. I reprinted the sensor lever using a .05mm layer setting and it works fine. I should have noted that the "0" reading for the IR sensor was wrong but didn't.
Thanks for all the help. My Bear Prusa MK3S Clone with a Bear frame and extruder appears to be working fine. Hope I do not have the need to bring problems with this clone back to the forum.
Two good points: I learned a lot more about my printer than I knew before; and, it now works as designed.
The extruder was much easier to work on with the BEAR build than the Prusa instructions to fix a similar problem on their new extruder body.
I assume that when I do my 2.5 to 2.5S upgrade (next) that I will be able to know all the different things I need to do to use the BEAR extruder.
I have printed a few extra .05mm layered levers so I will not have the same problem.

closed 2020-03-31 by gregsaun