gregsaun/prusa_i3_bear_upgrade #107: MK3S print failure in most Menu modes

taz43mca, 2020-03-24:
I have a MK3S with the Bear Frame  and Bear Extruder upgrades. Having worked through a few problems the printer does a great job BUT only if I put the menu on the LCD in the "Support...Sensor State" mode. If not in that mode, the printer does the 9 point bed check, goes to the start point and immediately goes to the right corner and gives me a "Filament remove" message, after doing that and replacing the filament, it returns to the left corner start point and repeats the same. 
I have checked all connections and even disconnected the filament sensor but same results, Also reflashed with latest FW, disabled the filament sensor, with same results. 
The Autoload and unload work as designed otherwise.
One person posted that they did a complete printer reset and the problem went away, not sure if they were talking about my particular problem, but I have not done that yet as I am not sure if I will then have to load the Bear FW and do an XYZ calibration, a complete wizard printer check, or whatever??? 
I also saw that some people had reported that some connections on their LCD to Enisy board had to be changed due to a problem.
I tried using Prontoface and the USB to print but got same results if the LCD is in any menu setting other than "Support...Sensor State".
Tried slicing on two different computers, using MK2.5, 2.5S, 3, 3S with same results except that the printer did recognize that I had not sliced in the 3S setting. I tried to print anyway and same results in all.
Finally, If I am printing in the "Support...Sensor State" menu mode and press the knob on the LCD, then the print stops and starts the crazy "filament removal process" repeatedly as before. I even tried to go back to the "Support...Sensor State" menu selection after the filament reload but that did not work to stop the cycle.

taz43mca, 2020-03-28:
Tried to completely reset the printer and use the BEAR FW to recalibrate all over again from scratch. Same results. Then flashed the Prusa 3.8.1 FW same results. I guess the person who did this and was successful was just lucky or maybe he did something I am not aware of. Printer is still humming along in the "Support...Sensor State" menu setting but no where else.

gregsaun, 2020-03-31:
Did you solved this issue? I am asking because other issue you wrote 😉

taz43mca, 2020-04-01:
All of my issues with the BEAR MK3 PRUSA clone I just finished a two color print with no problems (not using a MMU) using the Prusa slicer to set up color changes.All the help from many sources got me to this point. Hope my mistakes and problems help others when doing such a build. That includes GitHub, PRUSA, Ultimachine (both people and forums and procedures.Regards,Merrill AlburyStay healthy.

closed 2020-04-01 by gregsaun