gregsaun/prusa_i3_bear_upgrade #108: Calibration failed! Check the axes and run again

Idus, 2020-04-01:
So, I finally got the time to do my full Bear upgrade from a MK3s. I have been using the Bear x-ends and the Bear x-carriage for my MK3s Bondtech BMG extruder (https://www.thingiverse.com/thing:3477926) and made sure to get the extra official parts for upgrade 2.1 RC-1. 

I´ve now spent a couple of hours trying to fine-tune where the PINDA2 aims on the heatbed to get through the XYZ Calibration... but always getting "**Calibration failed! Check the axes and run again**".

However, I have realized something... If I run "Self Test" - all works out just fine. And as stated in step 20 in the latest RC-1 bear manual - the PINDA2 **IS** within the circle on the heatbed when it lowers during "Self Test". It is the exact same spot where the nozzle goes over the heatbed when I pick "Auto Home". (see pic.)

But if I run the "XYZ Calibration" (or Z "Calibration") - it moves to start in _another spot_ and stops before even continuing and failing... again.

Pinda2 have been calibrated to be "a cable-strap higher" than the nozzle as well as -0,5, +1, +1,5 and +2mm - all same results.

Note: Currently running latest Prusa FW 3.8.1 (and have tried with 3.9.0 beta))

**So, really stuck and don´t know what more to do?** 😕 Cannot find any solution in the manual or any forums. I´ve asked this in the Facebook-group as well - but no useful feedback.

[image]

gregsaun, 2020-04-01:
Hi Idus, you need to use the Bontech firmware:
https://support.bondtech.se/Guide/00.+Bondtech+Prusa+i3+-+Firmware+Update/44

Hope that helps!

taz43mca, 2020-04-01:
I had all the same problems. Used the suggested FW but unlike the directions, I did a complete system reset, flashed the Bear FW and then used the Wizard to do a complete calibration. After all done, then flashed with Prusa 3.8.1 FW for Mk 3S. Had one more unrelated problem but this fixed the "failed" problem. There is a long history of all this in the forum.

https://support.bondtech.se/Guide/00.+Bondtech+Prusa+i3+-+Firmware+Update/44

Idus, 2020-04-01:
> Hi Idus, you need to use the Bontech firmware:
> https://support.bondtech.se/Guide/00.+Bondtech+Prusa+i3+-+Firmware+Update/44
> 
> Hope that helps!

SUCCESS!! :-D You are the guru!

I have used original Prusa FW with the additional Bondtech M-commands for stepping (and the Bear x-ends/carriage) for over a year without any additional need for change anything else to calibrate etc, so never thought that the firmware needed to change when doing the full Bear.

All calibrated and working just fine - both smooth and textured plates.

Merci beaucoup and greetings from Sweden!

closed 2020-04-01 by Idus