gregsaun/prusa_i3_bear_upgrade #109: v2.1 Z Motor Mount - Shaft coupler clearance

trevjonez, 2020-04-07:
with the latest parts I am finding the clearance is just barely an issue.

older design looks to have a slimmer deck on the z axis (unconfirmed) compared to the 2.1 additionally the hole above the motor is smaller once past the nema 17 indexing step.

Issue is that on the newer design if you are using motors with a coupler clamp to drive your lead screws the hot end fan will bottom out on the coupler.

I have modified the parts for my own needs and that is literally what is printing in the image of the printer with 2.1 parts. So this is non blocking for me, but I thought you should know.

I opted to open the hole to make it work as the nema 17 alignment in the tapered opening and rod bottoming out on the motor relative to the 2040 seem to be the critical element. Certainly I have removed material so maybe deflection becomes an issue, but I believe it will be non critical.

here is my modified part: https://a360.co/2Xf1bkr
sketch 12, extrude 25, chamfer 19, and fillet 7 in that save are new edits from me. although the numbering is likely off due to me removing a few elements before making my changes. (uncofirmed, maybe numbering doesn't get affected by me deleting other features?)

old version of parts: 
[image]

newer version:
[image]

My specific coupler need only slide ~2mm lower to provide the coupler and fan enough clearance.

gregsaun, 2020-04-29:
Nice modification, I prefer to not add this on my parts as this is not a good idea to use coupler there. Also, the smaller center helps for stiffness so I prefer to keep it like that for now. However, could you share this somewhere and I add it to the optional parts from the community?

Gwyki, 2020-05-09:
@trevjonez do you not suffer with z-wobble nonsense with a coupler there? Your white parts do seem to show periodic z artefacts...

I came looking as I have had one of my screw integrated motors fail and was going to replace them with a coupler but ran into the clearance issue on my bear upgrade that you described. If it is a pain to tweak to get concentric, etc then I will just buy a new stepper! :/

trevjonez, 2020-05-10:
@Gwyki 
I have been fighting with z-wobble the whole time. biggest issue is just getting decent lead-screws.

I finally broke down and ordered some custom motors from LDO for my machines this last week.

You are spot on about the lines on my X ends. The interval on them is 2mm which coincides with the lead of the screw and is exactly the type of thing z wobble causes. Though with adjustment on the machine I was able to minimize it on future parts. but it comes back just about any time I adjust things and I have to repeat the effort of minimizing it.

@gregsaun 
Given the likely low usage of such a part I don't think I will upload the stl's to any distro site. I feel that anyone wishing to follow this path will either modify the parts on their own or find my F360 link in [the original post](https://github.com/gregsaun/prusa_i3_bear_upgrade/issues/109#issue-595599410)

And I totally agree it isn't ideal to have the coupler, but I wanted the higher resolution leadscrew to minimize backlash without resorting to spring based nuts. Custom motors will finally get me to the best of both worlds.

closed 2020-05-10 by trevjonez