gregsaun/prusa_i3_bear_upgrade #111: Y idler for 9mm belts step file not available

haleypearse, 2020-05-05:
My idlers are all 18mm in diameter whereas the cavity in the tensioner is ~17mm. If I had the step or fusion file I could easily make the required modification. Thanks.

gregsaun, 2020-05-05:
You should think about using a smaller idler later, those cheap idlers have too big flanges which can drag and wear the belt. I'll add step files in a second ;)

gregsaun, 2020-05-05:
Done :)

https://github.com/gregsaun/prusa_i3_bear_upgrade/commit/04889991a0d4c6dc647c66baf55d36db9cad4e24

haleypearse, 2020-05-06:
Awesome! Thanks. The fusion file was helpful. I can put up my version somewhere, maybe thingiverse, in case someone else has a cheap idler. My version is also worse because you have to thread belt through the tensioner before installing it, because the belt wants to go through the new back hole. But otherwise it seems fine.

I'll consider replacing it, or maybe modifying it, thanks. 

BTW even worse, I found I only had 20T pulleys so I had to model up a 16T 10mm one and print it this morning \_(:/)_/

I[image]

closed 2020-05-06 by haleypearse

gregsaun, 2020-05-06:
Thanks for the feedback and happy you could modify it. You can for sure add this to Thingiverse if you like, some people might be interested. However, I am not sure I want to add to list of optional parts from the community yet, I will see if I get this request regularly or not, for now you were the first one.