gregsaun/prusa_i3_bear_upgrade #112: No by gregsaun a beer button?

Bazmundi, 2020-05-20:
I am building a bear from scratch, using your great work.  Is there at least a buy a beer for the bear option?

gregsaun, 2020-05-20:
Hey, thanks for thinking about it, I really appreciate! Hope you will like it!

You can send me a tip via paypal: https://www.paypal.me/gregsaun
Or you can become my Patron: https://www.patreon.com/gregsaun

Thank you again :)

taz43mca, 2020-05-20:
I will add to that. I have built a Prusa i3 Mk 3S, Bear frame and extruder modification  clone 2.1 upgrade from scratch. Works great. Got quick response to my questions and missed steps. It all started out as a Bear frame upgrade planned for my Mk 2.5 but I got carried away. 2 printers are always better than 1 anyway.I understand Prusa may be coming out with a larger print bed model 2020-2021. I would have built my clone with a larger print area but didn't due to the non-availability of a larger magnetic bed, frame, & flex steel sheet. A little help with software/firmware & board modification would also be needed. To sum it all up...Gregson rocks.