gregsaun/prusa_i3_bear_upgrade #120: optional LCD cover does not fit the lcd supports (version 2.1)

domhardt, 2021-01-19:
The optional [LCD cover with text](https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/master/optional_parts/lcd_cover/printed_parts/stl/lcd_cover_text.stl?raw=true) has not enough space in the back to fit the [lcd_support_a](https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/master/optional_parts/lcd_cover/printed_parts/stl/lcd_support_a.stl?raw=true) and [lcd_support_b](https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/master/optional_parts/lcd_cover/printed_parts/stl/lcd_support_b.stl?raw=true). I refer to the version 2.1

gregsaun, 2021-01-20:
Hello, are you using the official Prusa LCD screen or an LDO Motor one?

domhardt, 2021-01-22:
Hello, as I got my printer from someone else I had to re-check everything. All dimensions of my parts are identical to the components from the Fusion360 project of this repo. So in the next step I assembled the components in Fusion360 and there is not enough space in the LCD cover for the LCD support brackets. Please see the attached images and the [Fusio360 project](https://a360.co/2LOxcg5).

[image]
[image]

gregsaun, 2021-01-23:
Oh I see, you are not using the correct mount. The custom cover comes with different mounts that are attached with 4 M3x10 screws. Check lcd_support_a and lcd_support_b here: https://github.com/gregsaun/prusa_i3_bear_upgrade/tree/master/optional_parts/lcd_cover/printed_parts/stl . (note the lcd_support_x_prusa is for original Prusa frame)

domhardt, 2021-01-23:
> Oh I see, you are not using the correct mount. The custom cover comes with different mounts that are attached with 4 M3x10 screws. Check lcd_support_a and lcd_support_b here: [...]

Oh, sorry. This is my fault then. Thank you for clarifying that. I will close this issue then. Sorry for bothering you.

Keep on the good work!

closed 2021-01-23 by domhardt