gregsaun/prusa_i3_bear_upgrade #122: Will there be a single-armed bear for prusa mini?

easonjim, 2021-08-03:
I saw that the prusa mini bear has been modified on the Internet, but it is a double-armed one, and the structural parts need to be repurchased. Can a prusa mini bear single-arm frame based on existing parts be produced?

gregsaun, 2021-08-03:
Those are project made by the community, I haven't made any Prusa Mini frame myself. Upgrading the Mini would not make a lot of sense to me as it would push it to price of MK3. I also don't have much time to work on another printer and if I was doing it would impact quality of my designs. For those reason I prefer to focus on the standard Bear only as of today.

easonjim, 2021-08-04:
Thank you for giving me ideas, I should focus on one product, not on as many models as possible.

gregsaun, 2021-08-04:
Happy if I can give some ideas, this is why I do open source! Btw, I like your profile image :)

closed 2021-08-15 by easonjim