gregsaun/prusa_i3_bear_upgrade #124: Update to MK3S+ Y Rod Holder

spooknik, 2021-09-13:
Hello! 

Just getting first getting into this project. I noticed you're still using the older style of Y-Rod holder with the Z-ties. Prusa has updated them for the MK3S+ to have screws. I don't see any functional difference but might look a bit nicer aesthetically. 

Do you have plans to update to the latest and would you welcome a PR if I had a go at it? 

Thanks for the cool project 👍 

Regards,
Spooknik

gregsaun, 2021-09-13:
Hi spooknik, I don't like those MK3S+ rod holders because you have to re-align the smooth rods everytime you have to do maintenance (on original MK3 as well). To me zipties are a better idea. You can find some mods on Thingi and co that are using two screws to maintain the rods if you prefer.

spooknik, 2021-09-13:
Thanks for the quick reply. Good point about the maintenance. I'll stick with the zipties for now. As I said for me it was mostly an  aesthetics thing. I can always "upgrade" later on if needed. 

Please close the issue if it suits you :)

closed 2021-09-13 by gregsaun

taz43mca, 2021-09-13:
I built one over a year ago. Used zip ties and have seen no reason to do otherwise.

Sent from the all new AOL app for Android