gregsaun/prusa_i3_bear_upgrade #126: MK3 Y carriage with endstop switch

Neukam, 2021-12-16:
I'm converting a MK1 style clone into what will essentially be a MK2.5S Bear using Prusa Firmware for Ramps.  I ordered a MK52 bed and MK3 Y carriage.  I will not have sensorless homing, so I will need to use endstop switches.  Will the MK2 motor/idler/belt/bearing mounts work with the MK3 Y carriage?

floridaservices, 2022-01-01:
My mk25s bear has end stops.  I have a rambo board but that should not make a difference.  I think you will just need to use the mk25 y motor mount, and if you are using the prusa mk3s+ parts for your x-axis just drill some holes in the left carriage piece to mount the x stop.  I have a mk52 12v and 24v bed,the mounting differences and bed support will not keep you from using endstops.  My mk25s started life as a mk1 with a wood frame and a mk2b heatbed

Neukam, 2022-01-01:
I've done some comparison on the DXF of both Y carriages and modified the MK3 belt holder to trigger the endstop.  I will be using the MK2 bearing holders.  I think I have a solution that will work correctly, but I haven't assembled and tested it yet.

floridaservices, 2022-01-01:
That is the difference I understand.  I have a 12V MK52 bed with the old style mounting so I was able to use the bear 2.5S y belt holder.  I don't do much design modification but it might be easier to modify the belt holder to fit the MK3 bed support, then you can use it with the 2.5S Y-motor holder.  I never figured out Prusa's reasoning for swapping the Y motor around but if you want to stick with the MK3 firmware you would have to do what you are planning but also modify the y-axis motor mount.

Neukam, 2022-01-01:
The plan is to use all MK2 mounts with a modified MK3 Y belt holder.  The modified belt holder has an extension added to trigger the endstop attached to the MK2 Y motor mount.  Assuming i calculated the offset correctly, and assuming the 24v bed aligns to the carriage in the same position as the 12v bed, the modified belt holder should work.

floridaservices, 2022-01-01:
I think it will work for you, it has been a while since I have had either of my printers apart that far but I seem to recall that the general location of the belt mount on both variations of the bed is the same, but they changed the screw spacing just enough.  I would work in some adjustment room if you can so you don't have to tweak firmware, unless you enjoy that sort of thing :)

floridaservices, 2022-01-15:
Came across this tonight, maybe this will help you: https://www.thingiverse.com/thing:3468567

Neukam, 2022-01-15:
I appreciate the input.  I have made a modified belt mount that should work with the MK3/MK52 carriage on a MK2 bear.  When i get a chance to finish the wiring and test it, I will report back and I'll post the modified part to thingiverse and link it here.

GMagician, 2022-02-14:
[y_belt_holder.zip](https://github.com/gregsaun/prusa_i3_bear_upgrade/files/8064184/y_belt_holder.zip)

These are what I used on my bear 2.1. I didn't use sensorless home not so reliable to me. And I readded switch. Please note that belt holder is mirrored compared to original bear 2.1 and motor support has to be mounted in a different position

Neukam, 2022-02-15:
I should have updated and closed this issue a while ago, but I had a hard drive die and lost the STL that I made.  I still have the GCode I used in Octoprint, so I'm currently reprinting the part.  I will measure the part that I am printing and recreate an STL so that I can upload it.  I feel guilty that I have held this issue open so long and I plan to close this issue tonight.  Thanks to all who have replied.

Neukam, 2022-02-15:
[y_belt_holder_MK3_Endstop.zip](https://github.com/gregsaun/prusa_i3_bear_upgrade/files/8065513/y_belt_holder_MK3_Endstop.zip)

I have uploaded the modified y_belt_holder above and also at prusaprinters.org.  [Here is the link](https://www.prusaprinters.org/prints/134527-prusa-bear-21-mk52-y-belt-holder-with-endstop).

closed 2022-02-15 by Neukam