gregsaun/prusa_i3_bear_upgrade #128: Suggestion: 10mm Ø Steel Rods

DiomedesDominguez, 2022-02-03:
First I'm not a hardware designer. Second, I was looking for any kit for a larger Y-axis, and found [this ones](https://orballoprinting.com/en/home/376-varilla-redondo-8mm.html) that are almost 200mm longer and then I though that maybe having such long rod will be easily bendable under the weight of the bed.

So, here is my suggestion: have a bigger diameter rods instead of the default 8mm ones.

gregsaun, 2022-02-08:
Hi, just to be sure I understand, are you going to use a longer  print bed as well?

DiomedesDominguez, 2022-02-08:
> Hi, just to be sure I understand, are you going to use a longer print bed as well?

yes.

gregsaun, 2022-02-08:
200mm longer print bed sounds excessive for bed sliding printers. Definitely 8mm smooth rids will have too much flex. It is hard to evaluate how stiff the 10mm should be but I would say it is going to be enough. You will also need a new Y carriage to support the bed.

DiomedesDominguez, 2023-07-08:
Here are the [bed ](https://orballoprinting.com/en/home/107-heatbed-mk3-aluminium-300x200-.html#/105-cama_caliente_12v_24v-24v) and the [Y carriage](https://orballoprinting.com/en/home/185-metal-eat-plate-carriage-y-axis-.html).