gregsaun/prusa_i3_bear_upgrade #134: Xyz calibration

gleny91, 2022-12-18:
When beginning to go probe 4 points saying check all axes.

gregsaun, 2022-12-18:
Hello, what extruder are you using?

gleny91, 2022-12-18:
I bought the one like the prusa

gleny91, 2022-12-18:
It came with the kit

gregsaun, 2022-12-18:
Is it an official Prusa kit?

gleny91, 2022-12-18:
No i bought it from aliexpress

gleny91, 2022-12-18:
Bought whole kit not upgrade

gregsaun, 2022-12-18:
Ok so we don't really know what it is. Could you share a picture of the X axis and extruder?

gleny91, 2022-12-18:
I am attaching ohotos

gleny91, 2022-12-18:
I don't know why they are not uploadingg

gleny91, 2022-12-18:
> ``

gregsaun, 2022-12-18:
you can drag and drop then in the issue directly. Otherwise you can use a site like Imgur

gleny91, 2022-12-18:
I am attaching photos but nothing is happening

gleny91, 2022-12-18:
Can we talk somewhere else ?

gleny91, 2022-12-18:
Could it be because it has original prusa firmware ?

gregsaun, 2022-12-18:
Yes if you have the Bear extruder this is why I wanted to see which one you have

gregsaun, 2022-12-18:
> Can we talk somewhere else ?

[here](https://guides.bear-lab.com/Wiki/Bear_FAQ?revisionid=HEAD#Section_How_could_I_contact_you)

gleny91, 2022-12-18:
Yes wherre ? Facebook ?

gleny91, 2022-12-18:
I fon't know why pics not uploading

gregsaun, 2022-12-18:
> I fon't know why pics not uploading

Use imgur

> Yes wherre ? Facebook ?

Discord, Twitter, Mastodon, Facebook and here

gleny91, 2022-12-18:
I sent a request on facebook

gregsaun, 2022-12-18:
I have answered on Facebook.

closed 2022-12-18 by gregsaun