gregsaun/prusa_i3_bear_upgrade #137: Kit build mk3s

MiguelitoHancho, 2023-03-30:
Hey Gregoire! When calibrating a new build mk3s/bear2.1/bearexxa  with the calibration firmware should I run a wizard calibration since it’s a fresh build and not upgrading an existing printer ? Also after completing the bear firmware and flashing back to prusa will I be able to run a full wizard calibration in the future or will it fail?  I don’t want to get that far and wind up having to do it all again haha  Thanks!!

Bartl1984, 2023-04-23:
Hi, unfortunately, I have the same problem. When I flash the latest Mk3s firmware 3.12.2, it demands a new Z calibration. This then fails. Calibration only works in Bear firmware, after flashing to Prusa, everything starts again from scratch. Do you have any tips? Thank you!

MiguelitoHancho, 2023-04-24:
Yea that same thing happened to me after trying the latest firmware .  I ended up using 3.12.1 and it has been fine.  It does ask you to calibrate it everytime you start it up , but doesn’t force you.  It seems like we can’t be the only ones but I feel like no one is addressing the issue haha.  Hope that helps 🤙

gregsaun, 2025-06-24:
As thisbis old and there is no more activity here I close this issue. Feel free to reopen in case of.

closed 2025-06-24 by gregsaun