gregsaun/prusa_i3_bear_upgrade #141: MK4 carriage on MK3S bearing holder

clicktoselect, 2023-07-08:
I had to get a new carriage for my MK3s and Prusa only sell the mk4 one now. The bearing holders aren’t quite big/deep enough to actually make contact with the carriage once the screws are tightened so they are sort of floating there. Has anyone else had this issue or found a solution? I know it’s a bit of a weird issue having mk4 carriage on mk3

gregsaun, 2023-07-08:
Hi @clicktoselect , thanks for reporting this, I was actually wondering how Prusa ship things regarding this. Did you receive the metal bearing clips with the carriage? Those in green in this page:
https://help.prusa3d.com/guide/2a-y-axis-assembly_406654#407495

clicktoselect, 2023-07-08:
It came by itself only, no bearing clips included sadly

gregsaun, 2023-07-08:
Looks like there is an issue with Prusa, they should have send you the bearing clips. Bearing clips have been introduced with MK3S+ which means if you are on MK3S you don't have those (it was U bolt at that time) and so the carriage they sell isn't compatible with MK3S. Might be good to talk to the support.

MK3S kit bed bearings assembly (steps 24 to 27): https://help.prusa3d.com/guide/2-y-axis-assembly_25488#26528
MK3(S) to MK3S+ upgrade (steps 4 to 7): https://help.prusa3d.com/guide/5-y-axis-upgrade_194395#194732 

On MK4 they have added bearing pads and plastic pads, I would ask the support if you could get those too. See the step 12 here:
https://help.prusa3d.com/guide/7-y-carriage-heatbed-assembly_438310#438578

I am not able to update those parts at the moment but I will update the assembly guide to explain that the new carriage should be used with the bearing clips.

Please let me know how it goes with the support :)

clicktoselect, 2023-07-08:
Thanks for your help Greg. I'll get in contact with Prusa support and see if they can offer me any of the new clips and pads. Thanks again!

gregsaun, 2023-07-08:
Good luck 👍

clicktoselect, 2023-09-25:
Just to follow up on this. I noticed the other day that the MK3s+ assembly manual now shows the new carriage, and does not actually require those pads that the mk4 has (https://help.prusa3d.com/guide/2a-y-axis-assembly_406654#408670)

In this case, it should be fine to just use the prusa bearing clips for the bear frame, instead of using the specific bear y bearing clips?

gregsaun, 2023-12-18:
I think the Bear version makes the Yvaxis a bit more silent. But both are fine.

sl1pkn07, 2025-05-10:
hi

i think prusa mixed up the instructions from MK4/S in the MK3/S, because few steps forward, still show the old Y carriage
https://help.prusa3d.com/guide/7-heatbed-psu-assembly_172872#173408

i think this because the new carriage needs the new hotbed spacers (https://www.prusa3d.com/es/producto/expansion-joints-set/) instead of the old ones (the hex spacers)

greetings

gregsaun, 2025-06-24:
As there is no more activity here I close this issue. Feel free to reopen in case of.

closed 2025-06-24 by gregsaun