gregsaun/prusa_i3_bear_upgrade #142: Tune/change filament from LCD-menu causes misalignment

FrHePo, 2023-11-23:
**Printer data:**
Printer type: MK3S
Printer firmware version: 3.13.2
Print via SD card
Bear parts: Lecktor Technologies OÜ - Värvi 5 C400-1 - 10621 Tallinn - Estonia
The Y-axis extrusion length is 331mm
All parts are original Bear.
The distance between the Y-rodholders is 311mm. The distance between the left side bearings is 93.8mm. This results in a max. build-plate travel of 217.2mm. The position of the Z-profiles was determined with the printed build_helper_z.stl-part (106 +/- 0.5mm). Mine measured 105.6mm.

**Problem description:**
When changing filament during printing with the LCD-function <Tune/change filament> the print head doesn't return to the correct position creating a Y-misalignment of about +0.7mm.
After I noticed this I did a cube test print (see picture) which confirmed me that this misalignment is independent of the printed object or the G-code. It's obvious that when the build-plate travels to the change filament position it runs into the Y-limit (by 0.7mm) which is actually an unintended collision.

![Uploading P1060828.JPG…
[image]
[image]

[image]
[image]
[image]

FrHePo, 2023-11-24:
@gregsaun My (outside/max.) distance between the left side bearings is 93.8mm. Could you check this please?
[image]

gregsaun, 2023-11-24:
Sorry I can't check CAD for noeàw but you can open the source file to check (Fusion 360).

I am wondering if the distance of the nozzle or the Pinda is closer to the X axis than what it should be. Where is this extruder coning from?

FrHePo, 2023-11-24:
The extruder is not involved in this problem. It's the build-plate that can't travel far enough (by only 0.7mm!) to reach the hard coded (firmware) filament change position. When calling the filament change by LCD-menu the build-plate is moved to the filament change position. My build-plate can't reach this position by 0.7mm. The left-side Y-bearing bangs into the Y-rodholder and can't travel any further.
My extruder is a combination of the Orbiter 2.0 and the E3D Revo Micro.

FrHePo, 2023-11-24:
3d-gussner stated: _We have defined [FILAMENTCHANGE_[XPOS|YPOS|ZADD] = X211 Y0 Z+2](https://github.com/prusa3d/Prusa-Firmware/blob/v3.13.2/Firmware/variants/MK3S.h#L359-L361) which are fine for our printers. And normally the Y axis can travel from [Y-4 up to 212.5] so for the stock printers it has some tolerances._
Standard MK3S Y-travel is [Y-4 up to 212.5]: 216.5mm
My max. Bear Y-travel is 217.2mm. So the max. Y-travel is not the problem. This leaves only 2 options for me: 1. The Y-bearings position is incorrect or 2. The position of the Z-profiles (frame) is incorrect and must be moved 5mm backwards.
The 1st option seems impossible to me. The Y-bearing-position cannot be incorrect by 5mm. That leaves only option 2 which means that the position of the Z-profiles (frame) is incorrect.

FrHePo, 2023-11-24:
Standard MK3:
[image]

My Extruder:
[image]

gregsaun, 2023-11-24:
You can check the dimensions of the frame here:
https://github.com/gregsaun/prusa_i3_bear_upgrade/blob/master/mechanical_drawings/bear21_frame_size.pdf

But Yes I suspect it is your extruder. Probably during calibration they stock somewhere the 0.0 point and would be different for you.

FrHePo, 2023-11-24:
Ok. I'll correct the Pinda position and hope and see :-))

FrHePo, 2023-11-26:
@gregsaun / @3d-gussner
After correcting the Y-Pinda-position by 3.5mm the misalignment has been cured. Thanks for your support! It made me a little wiser :-))
[image]

gregsaun, 2023-11-26:
Good news. I close it then.

closed 2023-11-26 by gregsaun